realsenseai/librealsense #3: "Sparse" alignment image upscaling?

floe, 2016-01-22:
Hi again, I'm sort-of misusing the issue tracker as a forum now, feel free to close if not intended :-)

That being said, I noticed that e.g. when using `rs::stream::depth_aligned_to_color` at a higher color resolution, you get a "sparse" image. This is, of course, expected as https://github.com/IntelRealSense/librealsense/blob/master/src/image.cpp#L356 only maps one single source pixel to a single target pixel.

In `libfreenect2`, we had implemented a naive-but-mostly-sufficient nearest-neighbor upscaling procedure for this case so that you can get a dense, color-aligned depth image. Would you be interested in a PR implementing a similar behaviour for `librealsense`? Then I'll prepare one.

ddiakopoulos, 2016-01-25:
Hi Florian, thanks for your pull request! This is indeed functionality that is desirable and fitting for inclusion in librealsense. Give us a couple days to work with our legal team on a CLA agreement for this project.

floe, 2016-01-26:
OK, let me know what you need. Instead of a full-scale CLA, would it be sufficient if I also declare my patch to be under Apache-2.0 license?

sgorsten, 2016-03-02:
Satisfied by pull request #9, merged to master in commit ed4bc8528cfe3853b7f8ac6e960c60f9597074ac.

Sorry for the long wait.

closed 2018-07-12 by dorodnic