realsenseai/librealsense #10: Segmentation fault when using multicam in 14.04.03

oliviakim95, 2016-01-25:
When connecting multiple cameras and runing multicam example file, I get the following output:

Segmentation fault (core dumped)
2016-01-25 13:23:45 INFO: UVC device detected with VID = 0x8086 PID = 0xa66
2016-01-25 13:23:45 INFO: Connecting to Intel RealSense F200
2016-01-25 13:23:45 INFO: UVC device detected with VID = 0x8086 PID = 0xa66

It seems like the devices aren't being registered separately.

ddiakopoulos, 2016-01-25:
Do you have an updated 4.4 kernel? Can you print out the firmware versions of the individual cameras?

oliviakim95, 2016-01-25:
Yes, I'm running the updated 4.4 kernel.
Do you know how I would check/update the firmware versions on linux?

ddiakopoulos, 2016-01-25:
cpp-tutorial-1-depth has a statement to print out firmware version. Also, could you check that librealsense can properly stream data from individual cameras?

oliviakim95, 2016-01-26:
I have one camera with 2.50.0.0, which is giving the segmentation fault, and one with 2.60.0.0. Do you know how I can update the firmware?

oliviakim95, 2016-01-26:
And also if there is a version after 2.60.0.0

sgorsten, 2016-01-26:
I've pushed a commit (bbf9e73a43eb0d38eb982dea2dd94cf8f55511b2) which should fix this issue.

The problem is that our Video4Linux backend was not correctly determining which USB device corresponds to which /dev/video* file. It was only looking for a VID/PID match, which meant that multiple instances of the same device on Linux would be incorrectly associated. The new code matches based on USB bus number and device number, which should work regardless of number and combination of devices are plugged in.

closed 2016-01-26 by sgorsten