realsenseai/librealsense #7: Install problem on ubuntu 14.04 "g++: error: unrecognized command line option ‘-mssse3’"

MEIXuYan, 2016-01-25:
Thank for your share !

When I install this on Ubuntu,I can't go with "make BACKEND=LIBUVC" ,it comes an error:

"g++: error: unrecognized command line option ‘-mssse3’".

I am new learner of realsense.Can you help me solve this ? Thank you.

ddiakopoulos, 2016-01-25:
What is your kernel version (`uname -r`) and your version of GCC (`g++ -v`)?

MEIXuYan, 2016-01-26:
The computer is not here currently,I will sent you the version information later.
I found the computer is with NVIDIA 4-Plus-1 (4-Plus-1™) ARM® Cortex-A15,not the Intel chip.
Is this the real problem?  If  so,do you know any mini-pc or embedded hardware platform(if it has to be intel product) can run realsense?
Because I really want to use the depth image on a small robot ,and it can be powered by battery.
Thank you very much.

MEIXuYan, 2016-01-26:
kernel version： 3.10.40
GCC version：4.8.4(Ubuntu/Linaro 4.8.4-2ubuntu1~14.04)

ddiakopoulos, 2016-01-26:
Hi. The `-mssse3` flag instructs GCC to enable SSSE3 x86 instructions which are required by a particular function in our library (yuy2=>RGB conversion in `image.cpp`). In order to compile without this flag for ARM architectures, you will need to re-write that functionality (a counterpart with NEON acceleration would be ideal, but a simple implementation would suffice). 

If you would like to move to x86, there are several products that might meet your needs for small and low power:

* Intel MinnowBoard Max 
* Kangaroo MD2B 
* Intel NUC
* The forthcoming Intel Compute Stick 2 (available in March 2016)

ddiakopoulos, 2016-01-26:
Also, if you need a reference implementation for yuyv to rgb conversion, here is a good source: https://github.com/ktossell/libuvc/blob/master/src/frame.c#L158

closed 2016-01-26 by ddiakopoulos

MEIXuYan, 2016-01-26:
Thank you for your time!

MEIXuYan, 2016-01-28:
I try to do yuyv to rgb conversion on ARM these day,but it is too hard for me as a beginner.
Since our robot just need the depth message(get 640X480 depth info array,our robot were based on Kinect One before),can I ignore the rgb data so that I can get depth data without using ssse3?
Is that a feasible scheme? Thank you.

bobdavies2000, 2016-01-28:
Yes.  You don't need to start or consume the RGB stream.

MEIXuYan, 2016-01-28:
Thank you. 
have you succeed in transfer it on ARM platform?
My Liunx kenral is v 3.10.40(is that suitable for librealsense?)
Currently I can't build the cpp-tutorial-1-depth.pro in qt,two errors occurs:
1.cstdio : no such file or dictionary
2.it shows no compiler,my gcc can't work (GCC version：4.8.4)