realsenseai/librealsense #12: Error while creating context sometimes after crash or debugging

Rennschnitzl, 2016-01-26:
When i debug my code or the program crashes i get one of the following error messages (seems to be random which i get) when i restart my code (or the example code)

rs_error was raised when calling rs_create_context(api_version:4):
    opcodes do not match

rs_error was raised when calling rs_create_context(api_version:4):
    calibration table is not compatible with this API

I can usually fix this by un- and replugging the usb cable.

ddiakopoulos, 2016-01-26:
Can you provide us some details about your system and camera? OS, kernel version, camera firmware, etc. Thanks!

Rennschnitzl, 2016-01-26:
OS: 
Ubuntu 14.04
3.19.0-47-generic #53~14.04.1-Ubuntu SMP Mon Jan 18 16:09:14 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

IDE / Compiler:
Qt Creator 3.6.0
Based on Qt 5.5.1 (GCC 4.9.1 20140922 (Red Hat 4.9.1-10), 64 bit)

Using device 0, an Intel RealSense F200
    Firmware version: 2.60.0.0

ddiakopoulos, 2016-01-26:
Hmm. We've had similar unplug/replug issues on the 3.19.xx series of kernels which is why our recommended configuration is 4.4 stable. If it is possible for your application, I would recommend reading over the installation guide and backing out of individual changes made in the `install-r200-udev-fix.sh` script. Then, upgrade to 4.4 per the guide and re-patch the uvcvideo driver.

Rennschnitzl, 2016-01-26:
Thanks for the info! Case dismissed.

I tried but since i couldn't install the nvidia drivers with 4.4, i went for 3.19.
Also it's F200, not R200.

closed 2016-01-26 by Rennschnitzl

ddiakopoulos, 2016-01-26:
One possible workaround for your configuration is to apply this during the uvcvideo patching process: https://patchwork.linuxtv.org/patch/25668/ but we haven't tested compatibility with the patch provided in this repo (line numbers may have changed which borks the ability to diff). https://patchwork.linuxtv.org/patch/25668/

ddiakopoulos, 2016-01-26:
One last thing! You might also experiment with the libuvc backend (semi-deprecated due to emphasis on v4l2 backend for Linux), but it's nice because no patches are required.