realsenseai/librealsense #13: Extended exposure times

JSewell-8, 2016-01-26:
Hi there.

I have been trying to use your library to capture individual images rather than for use as a stream. It looks as though the exposure is bounded from 0 to 1000 and that it is restricted by the frame rate. If for example, I wanted to take long exposure images. I have tried lowering the frame rate to 1-2 FPS with no success, there does not seem to be change. I also replaced the FPS check with '1' to see if I could get around that check. Is this a limitation of the hardware or of your library?

ddiakopoulos, 2016-01-26:
What camera are you trying to target?

JSewell-8, 2016-01-26:
The R200.

ddiakopoulos, 2016-01-26:
Thanks for the quick reply! Unfortunately it is a hardware limitation that the R200 does not enable single-shot or long exposure functionality (Actually it is theoretically possible to implement in firmware, but that's a different story...).

For F200 and SR300, there is a camera hardware control called Motion vs. Range tradeoff which will dramatically increase range (long exposure) at the expense of FPS.

While it might not be possible to enable your use-case now, it would be nice to know what application you need long exposure for so we can document and add to various future requirements lists.

floe, 2016-01-26:
I'll chime in here, too: I'm currently setting up an environment similar to HuddleLamp ([Website](http://huddlelamp.org/), [Paper](http://dl.acm.org/citation.cfm?id=2669500), [Video](https://www.youtube.com/watch?v=XkmwG588zp0)) with the F200, and the distance from camera to surface is ~ 80 cm. For this setup to work reliably, I've had to slightly increase the Motion/Range tradeoff.

JSewell-8, 2016-01-26:
Thank you for the quick response! This is helpful to know. I should mention it is the colour camera that I was trying to get a longer exposure on. 

My use case isn't very conventional so I can understand it isn't a highly desired feature. I just wanted to see how far I could use the sensor to take pictures as it is a pretty small device and would be convenient. I was planning to attempt HDR images.

sam598, 2016-01-26:
If motion isn't a concern, you should be able to take multiple video frames with a "full" exposure (33.33ms at 30fps) and add/average them together. This is essentially the same as taking a double exposure (adding gets you the added benefit that 2 frames would give you a 9 bit image, and 4 frames would give you a 10 bit image).

The only downside are possible motion artifacts in fast moving images. But it should work for a static scene.

JSewell-8, 2016-01-26:
I appreciate the suggestion. That was actually what I was doing before. I find that the resultant image is more underexposed than I would expect for good HDR.

ddiakopoulos, 2016-01-27:
Thanks for the discussion! Duly noted about the need for HDR support.

closed 2016-01-27 by ddiakopoulos