realsenseai/librealsense #16: Frame pointer ownership?

floe, 2016-01-28:
What's the policy regarding the frame data pointer returned from `rs_get_frame_data` https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense/rs.h#L390 ?
I.e., how long is it valid, who will delete it, ... ?

sgorsten, 2016-01-28:
The data returned by `rs_get_frame_data`, as well as `rs_get_frame_timestamp`, is valid until the next call to `rs_wait_for_frames`, or until `rs_stop_device` is called or the context itself is destroyed with `rs_delete_context`.

floe, 2016-01-29:
Thanks - perhaps a bit of background: I'm passing the buffers on to GStreamer, which then basically takes ownership of the pointer and would, at some point, try to free them again. Since I'll have to process the data anyway, I will probably just write it to a newly allocated buffer which GStreamer can then manage on its own.

closed 2016-02-04 by ddiakopoulos