realsenseai/librealsense #18: Creative Labs VF0800 support

johneleks, 2016-01-31:
Is Creative Labs VF0800 supposed to be supported at all? 

For now, USB vendor id 0x199e and product id are hardcoded 0x8101, while my device reports 0x041e (Creative Labs) and product id 0x4099.

After getting past it (by hardcoding these), cannot access control interface. 
uvc_scan_control then lists 4 interfaces:
0: bInterfaceClass = 1, bInterfaceSubClass = 1
1: bInterfaceClass = 1, bInterfaceSubClass = 2
2: bInterfaceClass = 1, bInterfaceSubClass = 2
3: bInterfaceClass = 3, bInterfaceSubClass = 0
I.e., first three are LIBUSB_CLASS_AUDIO and LIBUSB_CLASS_HID.
libusb_claim_interface returns -3 (LIBUSB_ERROR_ACCESS) for all interfaces except #2 (with or without sudo). Then, just to go further, interface #2 (audio!) lists no extension units. So here I am stuck.

If there any specification for VF0800 available?

My configuration: Mac OS X 10.9.5, MacBook Air 2013 (USB2.0, but behavior should not differ till data transfer is requested, correct?)

johneleks, 2016-01-31:
`bash-3.2$ ioreg -p IOUSB -b -w 0 -n "Creative VF0800@fd122000" 
+-o Root  <class IORegistryEntry, id 0x100000100, retain 9>
  +-o EHCI Root Hub Simulation@1A,7  <class IOUSBRootHubDevice, id 0x100000203, registered, matched, active, busy 0 (69 ms), retain 13>
  | +-o HubDevice@fa100000  <class IOUSBHubDevice, id 0x1000040c6, registered, matched, active, busy 0 (13 ms), retain 13>
  | | +-o BRCM20702 Hub@fa110000  <class IOUSBHubDevice, id 0x1000040ce, registered, matched, active, busy 0 (10 ms), retain 11>
  | | | +-o Bluetooth USB Host Controller@fa113000  <class IOUSBDevice, id 0x1000040d7, registered, matched, active, busy 0 (9805 ms), retain 12>
  | | +-o Apple Internal Keyboard / Trackpad@fa120000  <class IOUSBDevice, id 0x1000040d4, registered, matched, active, busy 0 (11550 ms), retain 14>
  | +-o FaceTime Camera (Built-in)@fa200000  <class IOUSBDevice, id 0x1000040cc, registered, matched, active, busy 0 (10829 ms), retain 13>
  +-o EHCI Root Hub Simulation@1D,7  <class IOUSBRootHubDevice, id 0x100000211, registered, matched, active, busy 0 (72 ms), retain 11>
    +-o HubDevice@fd100000  <class IOUSBHubDevice, id 0x1000040c0, registered, matched, active, busy 0 (13 ms), retain 13>
      +-o Card Reader@fd110000  <class IOUSBDevice, id 0x1000040cd, registered, matched, active, busy 0 (10965 ms), retain 9>
      +-o USB2.0 Hub@fd120000  <class IOUSBHubDevice, id 0x100004415, registered, matched, active, busy 0 (15 ms), retain 11>
        +-o Creative VF0800@fd122000  <class IOUSBDevice, id 0x10000441d, registered, matched, active, busy 0 (1573 ms), retain 13>
            {
              "sessionID" = 211130162133495
              "iManufacturer" = 1
              "bNumConfigurations" = 1
              "idProduct" = 16537
              "bcdDevice" = 4097
              "Bus Power Available" = 250
              "bMaxPacketSize0" = 64
              "USB Product Name" = "Creative VF0800"
              "iProduct" = 2
              "iSerialNumber" = 3
              "USB Address" = 5
              "bDeviceClass" = 0
              "bDeviceSubClass" = 0
              "locationID" = 18446744073660407808
              "IOUserClientClass" = "IOUSBDeviceUserClientV2"
              "PortNum" = 2
              "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
              "bDeviceProtocol" = 0
              "USB Vendor Name" = "Creative Technology Ltd"
              "Device Speed" = 1
              "idVendor" = 1054
              "USB Serial Number" = "201501143659"
              "Requested Power" = 50
              "IOGeneralInterest" = "IOCommand is not serializable"
              "Low Power Displayed" = No
            }
`

johneleks, 2016-01-31:
bash-3.2$ ioreg -p IOUSB -b -w 0 -n "Creative VF0800@fd122000" 
+-o Root  <class IORegistryEntry, id 0x100000100, retain 9>
  +-o EHCI Root Hub Simulation@1A,7  <class IOUSBRootHubDevice, id 0x100000203, registered, matched, active, busy 0 (69 ms), retain 13>
  | +-o HubDevice@fa100000  <class IOUSBHubDevice, id 0x1000040c6, registered, matched, active, busy 0 (13 ms), retain 13>
  | | +-o BRCM20702 Hub@fa110000  <class IOUSBHubDevice, id 0x1000040ce, registered, matched, active, busy 0 (10 ms), retain 11>
  | | | +-o Bluetooth USB Host Controller@fa113000  <class IOUSBDevice, id 0x1000040d7, registered, matched, active, busy 0 (9805 ms), retain 12>
  | | +-o Apple Internal Keyboard / Trackpad@fa120000  <class IOUSBDevice, id 0x1000040d4, registered, matched, active, busy 0 (11550 ms), retain 14>
  | +-o FaceTime Camera (Built-in)@fa200000  <class IOUSBDevice, id 0x1000040cc, registered, matched, active, busy 0 (10829 ms), retain 13>
  +-o EHCI Root Hub Simulation@1D,7  <class IOUSBRootHubDevice, id 0x100000211, registered, matched, active, busy 0 (72 ms), retain 11>
    +-o HubDevice@fd100000  <class IOUSBHubDevice, id 0x1000040c0, registered, matched, active, busy 0 (13 ms), retain 13>
      +-o Card Reader@fd110000  <class IOUSBDevice, id 0x1000040cd, registered, matched, active, busy 0 (10965 ms), retain 9>
      +-o USB2.0 Hub@fd120000  <class IOUSBHubDevice, id 0x100004415, registered, matched, active, busy 0 (15 ms), retain 11>
        +-o Creative VF0800@fd122000  <class IOUSBDevice, id 0x10000441d, registered, matched, active, busy 0 (1573 ms), retain 13>
            {
              "sessionID" = 211130162133495
              "iManufacturer" = 1
              "bNumConfigurations" = 1
              "idProduct" = 16537
              "bcdDevice" = 4097
              "Bus Power Available" = 250
              "bMaxPacketSize0" = 64
              "USB Product Name" = "Creative VF0800"
              "iProduct" = 2
              "iSerialNumber" = 3
              "USB Address" = 5
              "bDeviceClass" = 0
              "bDeviceSubClass" = 0
              "locationID" = 18446744073660407808
              "IOUserClientClass" = "IOUSBDeviceUserClientV2"
              "PortNum" = 2
              "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
              "bDeviceProtocol" = 0
              "USB Vendor Name" = "Creative Technology Ltd"
              "Device Speed" = 1
              "idVendor" = 1054
              "USB Serial Number" = "201501143659"
              "Requested Power" = 50
              "IOGeneralInterest" = "IOCommand is not serializable"
              "Low Power Displayed" = No
            }

johneleks, 2016-01-31:
Base on information in the article below, it seems that only USB 2.0 hub was shown. So this should be used with USB 3.0 machine only.

http://solsticlipse.com/2015/01/09/intel-real-sense-camera-on-linux.html

closed 2016-01-31 by johneleks

ddiakopoulos, 2016-01-31:
Hi -- the Creative Labs VF0800 "Senz3D" camera is not supported. I do believe it is actually a USB 2.0 device, but it is not a UVC compliant camera (you would need to poke some on-device registers and hack its isochronous streaming protocol). The Senz3D was a re-branded SoftKinetic ToF camera used with the original Perceptual Computing SDK and is no longer supported by the Intel RealSense team either.

teknotus, 2016-01-31:
> Hi -- the Creative Labs VF0800 "Senz3D" camera is not supported.

VF0800 is how the microphone USB audio shows up on the original F200. It's
a USB 2.0 device with a separate VID PID from the video.

ddiakopoulos, 2016-01-31:
Whoa! That is correct. I was thinking the Senz3D was the VF0800 not the VF0780. We almost always use the Intel names. In any case, neither the Senz3D nor the sidechannel microphone interface is supported by librealsense.