realsenseai/librealsense #19: Unable to control exposure of the color camera

taketwo, 2016-02-02:
I'm having troubles toggling auto exposure and getting/setting exposure time manually. `supports_option()` function works and reports that the `rs::option::color_enable_auto_exposure` is supported. However, when I try to get or set this option I get the following error:

```
RealSense error calling rs_get_device_option(device:0x19f7af0, option:COLOR_ENABLE_AUTO_EXPOSURE):
    VIDIOC_G_CTRL error 22, Invalid argument
```

At the same time, I can toggle auto exposure using `v4l2-ctl -d /dev/video1 -c exposure_auto=1`. So it seems to be a problem on the library side.

```
$ uname -a
Linux raccoon 3.19.0-49-generic #55~14.04.1-Ubuntu SMP Fri Jan 22 11:24:31 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

Tried with both R200 and F200. Any suggestions?

ddiakopoulos, 2016-02-10:
Hmm. We already have an inline comment for this as well https://github.com/IntelRealSense/librealsense/blob/master/src/uvc-v4l2.cpp#L459

It works for F200, R200, and SR300 on our Windows backend, so we'll look into it more deeply.

taketwo, 2016-02-12:
I'll also give it a try on my Arch laptop this weekend and let you know about the results.

digitologist, 2016-02-23:
Confirming this works fine on Windows but throws an error on Ubuntu 14.04.

ddiakopoulos, 2016-02-25:
UVC controls across the backends have slightly inconsistent interfaces when it comes to exposure. There is some additional logic that will need to be added to the V4L2 backend. Just noting here that we acknowledge color exposure is broken on Linux but at the moment it's lower priority to fix.

zhengye1995, 2016-04-05:
I aslo can notset this option in Window10，but I can set it，I used： 
dev->set_option(rs::option::color_exposure, 100);
but it is not worked

peci1, 2016-05-04:
I confirm this happens on 14.04 when using the Realsense ROS package and setting R200_LR_AUTO_EXPOSURE_ENABLED to 1. That makes the IR and depth images stall until I reconnect the device.

What's also interesting: the cpp-enumerate-devices example shows the following:

    COLOR_ENABLE_AUTO_EXPOSURE : 0 .. 0, 1

Which I understand as the color auto exposure having range 0-0. That's strange.

taketwo, 2016-07-14:
As I noted before, it is possible to toggle auto exposure and adjust exposure time using `v4l2-ctl` command-line tool. I had a look inside the source code, and it turns out they use different control IDs. For auto exposure it is `V4L2_CID_EXPOSURE_AUTO`, and for exposure time it is `V4L2_CID_EXPOSURE_ABSOLUTE`.

So this patch fixes the issue for me:

```patch
diff --git a/src/uvc-v4l2.cpp b/src/uvc-v4l2.cpp
index 9c1b01b..05a94f5 100644
--- a/src/uvc-v4l2.cpp
+++ b/src/uvc-v4l2.cpp
@@ -462,14 +462,14 @@ namespace rsimpl
             case RS_OPTION_COLOR_BACKLIGHT_COMPENSATION: return V4L2_CID_BACKLIGHT_COMPENSATION;
             case RS_OPTION_COLOR_BRIGHTNESS: return V4L2_CID_BRIGHTNESS;
             case RS_OPTION_COLOR_CONTRAST: return V4L2_CID_CONTRAST;
-            case RS_OPTION_COLOR_EXPOSURE: return V4L2_CID_EXPOSURE; // Is this actually valid? I'm getting a lot of VIDIOC error 22s...
+            case RS_OPTION_COLOR_EXPOSURE: return V4L2_CID_EXPOSURE_ABSOLUTE;
             case RS_OPTION_COLOR_GAIN: return V4L2_CID_GAIN;
             case RS_OPTION_COLOR_GAMMA: return V4L2_CID_GAMMA;
             case RS_OPTION_COLOR_HUE: return V4L2_CID_HUE;
             case RS_OPTION_COLOR_SATURATION: return V4L2_CID_SATURATION;
             case RS_OPTION_COLOR_SHARPNESS: return V4L2_CID_SHARPNESS;
             case RS_OPTION_COLOR_WHITE_BALANCE: return V4L2_CID_WHITE_BALANCE_TEMPERATURE;
-            case RS_OPTION_COLOR_ENABLE_AUTO_EXPOSURE: return V4L2_CID_AUTOGAIN; // Automatic gain/exposure control
+            case RS_OPTION_COLOR_ENABLE_AUTO_EXPOSURE: return V4L2_CID_EXPOSURE_AUTO;
             case RS_OPTION_COLOR_ENABLE_AUTO_WHITE_BALANCE: return V4L2_CID_AUTO_WHITE_BALANCE;
             default: throw std::runtime_error(to_string() << "no v4l2 cid for option " << option);
             }
```

Tested with all (R200, F200, SR300) cameras. Note that `V4L2_CID_EXPOSURE_AUTO` encodes modes with enum, where 1 is manual mode, and 3 is aperture priority. This means that to disable auto exposure one needs to call `device->set_option(rs::options::color_enable_auto_exposure, 1)`, and to enable it `device->set_option(rs::options::color_enable_auto_exposure, 3)`.

taketwo, 2016-08-26:
I've been using `librealsense` with this patch for over a month now on Ubuntu 14.04 and 16.04 without any problems. Should I submit a pull request?

dorodnic, 2016-08-26:
Thank you, this does seem to solve the problem. 
This fix is also available on the development branch and will be merged to master.

dorodnic, 2016-09-06:
Auto exposure should be functional now on all platforms

closed 2016-09-06 by dorodnic