realsenseai/librealsense #21: Unable to build librealsense on Nvidia Jetson TK1

mtsakaguchi, 2016-02-04:
Hello,

I am trying to install the drivers to get the RealSense R200 working on my Jetson TK1 board (ARM processor) which runs Ubuntu14.04 (Linux4Tegra). I'm having issues patching the uvcvideo.ko file. For the Video4Linux backend, when I run the "patch-uvcvideo-4.4.sh v4.4-wily" it hangs on when cloning the repository linux-v4.4-wily even though I've let it sit for a long period of time. In addition, when I try and use the "patch-ucvvideo-3.19.sh" script the patch cannot be applied because apt-get cannot retrieve a linux-image (which I believe is due to the fact that the Nvidia Linux4Tegra kernel is only available for download through the Nvidia site).

I've also tried the LibUVC backend but when I go to build "make BACKEND=LIBUVC" I get the error:

mkdir -p obj/libuvc
mkdir -p lib
mkdir -p bin
g++ src/context.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_LIBUVC_BACKEND `pkg-config --cflags --libs libusb-1.0` -c -o obj/context.o
g++: error: unrecognized command line option '-mssse3'
make: *** [obj/context.o] Error 1

Any help would be greatly appreciated!

teknotus, 2016-02-04:
> I am trying to install the drivers to get the RealSense R200 working on
my Jetson TK1 board (ARM processor) which runs Ubuntu1404 (Linux4Tegra)

Currently librealsense won't build for ARM. It has some hand tuned intel
assembly for example. I've personally used the F200 on a Jetson TK1. It
seemed a little slow, but I didn't write any GPU code. It's on the low end
of what can handle the data rate. The required kernel patch is very similar
for R200 so I'm fairly certain it can be done. I partially automated the
kernel update process for F200.

https://github.com/teknotus/jetson-grinch-installer

ddiakopoulos, 2016-02-04:
You should reference issue #7. All of the existing scripts will work on standard x86 Ubuntu machines only. It is a good idea to run with the libuvc backend, but as noted in #7 and by @teknotus, you need to replace the yuyv=>rgb conversion routine (look back at the commit log for an implementation before we accelerated with SSE). The actual you are facing above is related to the fact your GCC compiler on ARM has no idea what SSSE3 instructions are -- you need to remove that flag and replace the conversion routine.

closed 2016-02-04 by ddiakopoulos

ukamhou, 2016-03-04:
Hi ddiakopoulos,

I am also trying to get this compiled on Tegra X1 processor, can you be more specific as for which commit log I should go back to? I tried searching for SSE in the commit log but couldn't find a match.

Also, if anyone has successfully modified the implementation, can you please post instructions in doing so. I think there are many people who would be interested in getting this to work on ARM platform.

Thanks in advance!