realsenseai/librealsense #22: cpp-multicam std::runtime_error at memory location...

glang, 2016-02-06:
After importing all the example projects into Visual Studio Profession 2013 with update 5 on Windows 10, all the example projects run fine, besides cpp-multicam, which crashes immediately.

My error output is shown below, and I see that toward the bottom, there are runtime_errors...

'cpp-multicam-d.exe' (Win32): Loaded 'C:\Users\Granger Lang\Desktop\test\librealsense-master\bin\Win32\cpp-multicam-d.exe'. Symbols loaded.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\ntdll.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\kernel32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\KernelBase.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Users\Granger Lang\Desktop\test\librealsense-master\bin\Win32\glfw3.dll'. Module was built without symbols.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\opengl32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Users\Granger Lang\Desktop\test\librealsense-master\bin\Win32\realsense-d.dll'. Symbols loaded.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\msvcrt.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\msvcp120d.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\advapi32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\sechost.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\msvcr120d.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\rpcrt4.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\sspicli.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\cryptbase.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\bcryptprimitives.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\user32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\ole32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\combase.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\gdi32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\shlwapi.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\glu32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\ddraw.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\setupapi.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\cfgmgr32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winmm.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\msvcr120.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\mf.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\mfplat.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\mfreadwrite.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\dciman32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winmmbase.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winmmbase.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winusb.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winmmbase.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\winmmbase.dll'
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\winmmbase.dll'
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\mfcore.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\ksuser.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\mfperfhelper.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\imm32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\kernel.appcore.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\RTWorkQ.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\uxtheme.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\oleaut32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\clbcatq.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\deviceaccess.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\vidcap.ax'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\Kswdmcap.ax'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\mfc42.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\WinSxS\x86_microsoft.windows.common-controls_6595b64144ccf1df_5.82.10586.0_none_811bc0006c44242b\comctl32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\odbc32.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\dpapi.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\policymanager.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\msvcp110_win.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\devobj.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\wintrust.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\msasn1.dll'. Cannot find or open the PDB file.
'cpp-multicam-d.exe' (Win32): Loaded 'C:\Windows\SysWOW64\crypt32.dll'. Cannot find or open the PDB file.
First-chance exception at 0x75CDD8A8 in cpp-multicam-d.exe: Microsoft C++ exception: std::runtime_error at memory location 0x0018EA88.
First-chance exception at 0x75CDD8A8 in cpp-multicam-d.exe: Microsoft C++ exception: [rethrow] at memory location 0x00000000.
First-chance exception at 0x75CDD8A8 in cpp-multicam-d.exe: Microsoft C++ exception: rs::error at memory location 0x0018F6B0.
First-chance exception at 0x75CDD8A8 in cpp-multicam-d.exe: Microsoft C++ exception: std::runtime_error at memory location 0x02B8DCAC.
The thread 0x1cd4 has exited with code 0 (0x0).
First-chance exception at 0x75CDD8A8 in cpp-multicam-d.exe: Microsoft C++ exception: std::runtime_error at memory location 0x02C8DEA8.
The thread 0x1d08 has exited with code 0 (0x0).
The thread 0x1cf4 has exited with code 0 (0x0).
The thread 0x1ce0 has exited with code 0 (0x0).
The thread 0x1d68 has exited with code 0 (0x0).
The thread 0x1d58 has exited with code 0 (0x0).
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\deviceaccess.dll'
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\dpapi.dll'
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\odbc32.dll'
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\mfc42.dll'
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\WinSxS\x86_microsoft.windows.common-controls_6595b64144ccf1df_5.82.10586.0_none_811bc0006c44242b\comctl32.dll'
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\Kswdmcap.ax'
'cpp-multicam-d.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\vidcap.ax'
The thread 0x1cd0 has exited with code 1 (0x1).
The thread 0x1d34 has exited with code 1 (0x1).
The thread 0x1cc8 has exited with code 1 (0x1).
The thread 0x1d10 has exited with code 1 (0x1).
The program '[1372] cpp-multicam-d.exe' has exited with code 1 (0x1).

ddiakopoulos, 2016-02-06:
In one of the Visual Studio menus there is a window where you can enable breakpoints on C++ exceptions. Can you turn that on and report back? It will give you the exact location of the runtime_error. Also, the console output in most (all?) examples should catch the runtime error and print a helpful message.

glang, 2016-02-06:
I enabled breakpoints for thrown exceptions and ran the multicam program again. Here is the first thing that pops up: [screenshot](http://i.imgur.com/zHVo7q1.png)

Is there anything in particular that you would like to see?

ddiakopoulos, 2016-02-06:
Our bad. If you look at this line https://github.com/IntelRealSense/librealsense/blob/master/examples/cpp-multicam.cpp#l31, we accidentally left a hardcoded streaming mode that only works for the R200 camera. Just replace the lines with the commented versions and it should work, and we'll fix it up in the actual example Monday!

glang, 2016-02-06:
I see! Thanks! 

On a unrelated note, I got an email from SFVR saying that you'll be giving a talk at their next event! I'm a huge VR enthusiast and was wondering if you know any good resources for learning how to integrate the RealSense cameras with VR headsets?

ddiakopoulos, 2016-02-07:
Come to the event to find out!

closed 2016-02-07 by ddiakopoulos

glang, 2016-02-07:
I would! But I'm an undergrad attending a university that's a few hours away, unfortunately :/

glang, 2016-02-09:
Hey @ddiakopoulos , so there is no longer an error for when cameras are connected. However, an runtime exception is still thrown when there are no cameras connected. Just though I should let you guys know that.

ddiakopoulos, 2016-02-10:
@glang -- maybe I'll type up a blog post about it after the event!

On your second note, librealsense can and will throw exceptions as an error handling mechanism. It is by design that the examples will throw when there are no connected cameras.

glang, 2016-02-10:
@ddiakopoulos That would be awesome! And thanks for fixing the camera issue!

glang, 2016-02-21:
@ddiakopoulos Now that's the SFVR event is over, would you be able to share what you presented? I'm really curious!