realsenseai/librealsense #23: Windows 10 Pro N Mf.dll dependency

Jason1820, 2016-02-07:
Hi,

I have used the library on a 64-bit windows 10 machine running VS 2013 community. This works perfectly. However, when I try on another device which is a 64-bit windows 10 Pro N machine I get messages for 'cannot find MF.dll MFPlat.dll and MFReadWrite.dll'. I am not sure if this is an issue with the library or just with my configuration. 

If there is a dependency on these dlls I can turn off to help, that would be greatly appreciated information to know.

bobdavies2000, 2016-02-07:
Install : Media Feature Pack for N and KN versions of Windows 10 :
https://www.microsoft.com/en-us/download/details.aspx?id=48231

Jason1820, 2016-02-07:
Thank you for the suggestion. I tried this but it said I had already had it installed.

ddiakopoulos, 2016-02-07:
This is not an issue with librealsense, but a rather well documented missing component of Windows N installations. You will need to debug why the Media Feature Pack is not installing on your configuration. It contains the media foundation dlls which are critical to interacting with UVC camera devices as the OS layer.

closed 2016-02-07 by ddiakopoulos

bobdavies2000, 2016-02-07:
You could try to uninstall it and reinstall the Media Feature Pack.  The
update contains the missing DLL's.  You need to make sure that you have
both the 32-bit and 64-bit versions.  They should reside in
c:\windows\syswow64 (64-bit) and c:\windows\System32.

gknapp, 2018-02-13:
There are many versions of Media Feature Pack now. Check you install the version that corresponds to your build, otherwise the installer won't copy these DLLs in place.

https://support.microsoft.com/en-us/help/3145500/media-feature-pack-list-for-windows-n-editions

meniiu3210, 2023-09-21:
windows 10 pro n