realsenseai/librealsense #24: Multi-cam with USB hub

Jason1820, 2016-02-08:
I have been trying to connect multiple cameras to a USB 3.0 hub. I notice I can stream 2 streams with the highest colour resolution but any more than that and the library crashes. I am not too concerned about the frame rate, if it is slow that is not a big deal. Has anyone had any success with this?

bobdavies2000, 2016-02-08:
I have tested it with 3 cameras on my desktop running Windows 8.1 and
Windows 10.  What is your setup like?

Jason1820, 2016-02-08:
Windows 10. Did you test it with the 3 cameras all connected to the same USB hub? I can get 3 streams running if I connect 2 to the hub, then another to a 2nd USB port on my computer but I would prefer for them all to use the same hub.

digitologist, 2016-02-08:
Is your hub bus-powered or does it have a supplementary power supply? Beyond two cameras per hub, I have had better luck with powered hubs, but still find it safest to limit to two. I can get eight cameras working on a NUC no problem, but need to use 4 hubs to do it.

Jason1820, 2016-02-08:
Powered USB 3.0 hub. This has been tested with 2 different hubs.

ddiakopoulos, 2016-02-08:
As @digitologist states, our recommended config is 2 cameras per hub. The 1920x1080 rgb stream is 6 megabytes (x2 streams x15 times a second) + depth -- it ends up saturating most USB 3.0 hub chips quite quickly.

teknotus, 2016-02-08:
I know with people using multiple kinects the answer was getting a USB card
that had a separate controller for each port. Most cards have one
controller and a USB hub. Anyone know of a USB 3 interface like that? PCIe
for a NUC or Thunderbolt 3 for a laptop.

Jason1820, 2016-02-09:
The controller would switch on and off different ports if I am right? That seems to be one solution, to capture from 2 camera, then switch to the other two. Although it is not so convenient.

ddiakopoulos, 2016-02-10:
There are mini-pcie controllers that can add additional USB 3.0 ports to a NUC (but they aren't convenient -- you basically have to drill a hole in the case due to space constraints on the breakout). Not sure about the case for Thunderbolt 2/3.

Closing this issue out because we've stated the recommended/known working configuration of 2 cameras per powered USB 3.0 hub.

closed 2016-02-10 by ddiakopoulos