realsenseai/librealsense #26: TemperatureControlLoop error when using multiple cameras

oliviakim95, 2016-02-09:
I am getting the following error:

rs.error: TemperatureControlLoop: opcodes do not match

when I try to connect multiple f200s.

ddiakopoulos, 2016-02-10:
Have you updated your build from the master branch recently?

oliviakim95, 2016-02-10:
Yes, I have the latest changes, but I have first noticed this since commit number 86efd7cc512d58aeeba92115b8247802a2c9ab9a

Wollimayer, 2016-02-10:
Do you stop the camera device after taking an Image ? I had the same problem after I tried to start/stop each camera after a frame was captured. (recent build)

oliviakim95, 2016-02-10:
Nope, never made a call to device stop

Wollimayer, 2016-02-10:
Did you check your cpu usage ? The current version seems to "crash" ( actualy it freezes) if there is too much stuff going on in the background. (That's why i tried to stop the camera after taking a frame)

oliviakim95, 2016-02-10:
I did check it, it isn't very high - I only stream from one device at a time and am still getting this error.

ddiakopoulos, 2016-02-10:
So the TCL runs every 10 seconds. Do you see this error consistently every 10 seconds or simply on occasion? On occasion isn't necessarily an error (even though we report it as such).

oliviakim95, 2016-02-10:
It is consistently reoccuring

ddiakopoulos, 2016-02-10:
Based on the past two issues you've filed I'm beginning to suspect you might have a damaged/broken camera. I see that you're on the latest Ubuntu 14.04 + 4.4 release and a recent firmware version which passes our validation for F200 (i.e. no streaming timeouts for a single cam and minimal temp control loop failures). If you ordered a devkit directly through Intel, you can request a replacement by emailing click.support@intel.com.

ddiakopoulos, 2016-03-10:
Closing for lack of new information.

closed 2016-03-10 by ddiakopoulos