realsenseai/librealsense #27: Recovering from timeout error (question)

digitologist, 2016-02-10:
Running my app with 8 R200s, I found that individual cameras would start throwing timeout errors.  The error handling allowed me to recognize and skip the future wait-for-frames call for the affected devices, but I wasn't able to find the correct way to recover the error without restarting the application.  

This was a live installation so I didn't have the opportunity to stop & restart the app, let alone unplug/replug the device, so my number of active cameras simply dropped over time.  

Is there a correct way to allow for a camera that's not returning frames to be closed, unplugged/replugged, and then recover streaming?

ddiakopoulos, 2016-02-10:
For R200 we have a few unpublished commits that let you reset the device after a frame timeout. Ping @sgorsten to push them in the next few days.

Wollimayer, 2016-02-10:
How do you catch the timeout errors? I do use try-catch to wrap the code for wait-for-frames and frame capturing, but it looks like no error is thrown. The programm freezes while waiting for a frame and never gets back.

ddiakopoulos, 2016-02-10:
@digitologist do you have a snippit of code to share with @Wollimayer?

digitologist, 2016-02-10:
Here's how I do it:
```
for (int i = 0; i < deviceCount; ++i) {
	if (deviceAbort[i]) continue;
	try {
		deviceManager[i]->wait_for_frames();
	}
	catch (const rs::error & e) {
		deviceAbort[i] = true;
		cout << "Realsense Error: cam " << i << ": " << e.what() << endl;
	}
}
```

Wollimayer, 2016-02-11:
@digitologist Thank you. I'm using the same construct right now, but no error is thrown. 
It might be a different problem ( issue #29  ).

ddiakopoulos, 2016-02-11:
@Wollimayer - can you use one of the other sample apps to print out the firmware versions of each of your F200s?

Wollimayer, 2016-02-13:
2.60.0.0 And two 2.50.0.0 . 
The problem occurs on both versions.

ddiakopoulos, 2016-02-13:
@Wollimayer will discuss in #29, streaming timeouts and config timeouts are logically different issues. 

For streaming timeouts, code on the the current dev branch should be able to reset a device fully at runtime after code like @digitologist detects a timeout.

closed 2016-02-13 by ddiakopoulos

MFUHH, 2017-06-29:
Hi, I also have problems with timeout of streaming. I can catch the error, stop and start the camera and then the stream is running. But the timestamps then are wrong. the get_frame_timestamp function gives numbers between -3 and 4. Is that a bug? How can I restart the camera properly?