realsenseai/librealsense #30: ROS interface

ffurrer, 2016-02-15:
Is there already any plan to create a full functionality ROS interface for this?

ddiakopoulos, 2016-02-15:
Absolutely! We have an internal team working on exposing all librealsense functionality as a ROS node. It's trending toward an early March release, and will be put on this same Github (github.com/IntelRealSense). I'll write a note here when it's been posted so you get a notification when it's available.

ffurrer, 2016-02-15:
awesome, thanks :)

ibaranov-cp, 2016-02-26:
*Poke* Any news? :)

y22ma, 2016-02-26:
Yo @ibaranov-cp, feel free to use my wrapper in the meantime here. https://github.com/y22ma/realsense_multi_cam

Yan

ddiakopoulos, 2016-02-26:
Really soon™. It's finished but going through internal validation + legal.

ddiakopoulos, 2016-03-01:
:boom: :ok_hand: https://github.com/intel-ros/realsense

closed 2016-03-01 by ddiakopoulos