realsenseai/librealsense #32: Build without glfw3

vidstige, 2016-02-16:
It would be awesome to be able to run without glfw3 as I don't care about builtin visualizations. It would be great if that was more put inside examples or such.

ddiakopoulos, 2016-02-16:
The core librealsense library does not depend on or link against glfw. Several projects (c/cpp-tutorial-1 and cpp-headless) specifically do not include glfw3 for the reason you list above. Hope that helps?

vidstige, 2016-02-17:
Alright, sounds great. That's the most important part. Is there also a way to build without having glfw3 installed using the makefile?

ddiakopoulos, 2016-02-17:
Good question. The makefile would require a boolean flag similar to the backend selection so that it would only compile the examples that do not need glfw. This would require enumerating the .c/.cpp files exactly (like cpp-headless.cpp instead of the current wildcard *.cpp), which is slightly harder to maintain. This probably won't be on our roadmap right away (for headless systems we've been fine manually editing the makefile), but would accept a pull request if you were to make and test the relevant changes. All told, should be about ~10 lines of addition to the makefile.

vidstige, 2016-02-18:
Cool. Thanks for a quick answer. Just have to say after using the Windows SDK, a download of a few GBs anda bit bloated. I'm super impressed by this software. <3

closed 2016-02-18 by vidstige

vidstige, 2016-02-18:
For future reference: It is _already_ possible by simply issuing 

    make library