realsenseai/librealsense #33: Using make all results in missing file: 'linux/usb/video.h'

eyalfoni, 2016-02-16:
I have followed all the download instructions but when I try to build examples, I am missing the above file. Please let me know what I can do to fix this.

ddiakopoulos, 2016-02-16:
This is not necessarily a librealsense issue -- but rather an issue with your system configuration. For some reason you are missing the V4L/uvcvideo headers on your system. What distribution are you using, and what kernel version? Have you successfully patched the uvcvideo.ko driver?

eyalfoni, 2016-02-16:
Distribution:
Darwin Kernel Version 15.2.0: Fri Nov 13 19:56:56 PST 2015; root:xnu-3248.20.55~2/RELEASE_X86_64 x86_64
Where is the patch for the uvcvideo.ko driver?

ddiakopoulos, 2016-02-16:
Ok, so you're actually on OSX! According to our CI build, the XCode project should be building A-OK. If you do want to use the makefile (which it appears you are attempting to use) check out this pull request: https://github.com/IntelRealSense/librealsense/pull/17 The makefile was originally only for Linux systems but #17 might fix your issue

blackerpaper, 2016-02-17:
I'm using Xcode and opening librealsense.xcworkspace. All the example compiled and running fine. Maybe you use the standard Makefile just like in linux. It didn't work in osx. Use the provide Xcode project to compile it.

I make the cmake in my fork, if you prefer using cmake.

Xcode 7.2 elCapitan

eyalfoni, 2016-02-17:
Thanks for the advice! I got it working in Xcode by using the librealsense.xcworkspace!

closed 2016-02-17 by eyalfoni