realsenseai/librealsense #36: Segfault when F200 has non-sequential video device numbers (Linux)

floe, 2016-02-17:
I noticed today that all my custom code, as well as all examples, suddenly started segfaulting.
The reason: due to some previous camera plugging and unplugging, the two video devices exposed by the F200 were assigned to `/dev/video2` and `/dev/video4`. Once I removed the camera claiming `video3` and re-plugged the F200, everything worked again.

I'm aware this is a very special scenario, just thought I'd mention it here in case somebody else runs into this.

ddiakopoulos, 2016-02-17:
Nice find. I don't think we ever saw a case where individual devices had non-sequential handles. Here is how we group subdevices: https://github.com/IntelRealSense/librealsense/blob/master/src/uvc-v4l2.cpp#L524. libusb has a similar issue (how do you associate the IF with the device) which we fixed by matching device number on a bus (https://github.com/IntelRealSense/librealsense/blob/master/src/uvc-v4l2.cpp#L546) Looks like we'll need to go implement the same mechanism for streaming subdevices.

floe, 2016-02-17:
As I said, it's a very special scenario (have at least 2 cameras present, remove the first one, _then_ connect the F200), so I'm not sure if it needs a code fix. Maybe a note in the FAQ is enough.

sgorsten, 2016-03-02:
Fixed in b8a83dfcf606cc129a0dacb86288147e6a97e3cd.

V4L2 video devices are now explicitly grouped by USB busnum/devnum, and sorted by multiple interface index.

closed 2016-03-02 by sgorsten