realsenseai/librealsense #40: SR300 enable_stream issue

danfeiX, 2016-02-23:
It appears to be that there is no other preset available other than best_quality for the SR300 device. Is the support for SR300 going to be released soon?

ddiakopoulos, 2016-02-23:
SR300 is fully supported! It might have been easy to miss, but the stream presets are only provided for convenience. You are be able to set pixel format, framerate, and resolution independently using a similar function: https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense/rs.hpp#L306

You can quickly see some of the supported modes here: https://github.com/IntelRealSense/librealsense/blob/master/src/f200.cpp#L123

closed 2016-02-23 by ddiakopoulos

danfeiX, 2016-02-24:
Hi Thanks for the quick response! 
However, I get "unsupported stream" exception when I enable the stream with 
dev->enable_stream(rs::stream::color, 1920, 1080, rs::format::any, 30);

ddiakopoulos, 2016-02-24:
try rs::format::rgb8