realsenseai/librealsense #44: Is it possible to run the library without having root permissions?

Kukanani, 2016-02-26:
For example, running `cpp-pointcloud` results in the error:

`RealSense error calling rs_create_context(api_version:4):
    libusb_open(...) returned LIBUSB_ERROR_ACCESS`

I see from other forums that the solution to this is to add myself to the root group or run with 'sudo', but I'd prefer not to have to do this on every machine on which we want to use the library. My lab has several different hardware platforms and this would be a pain in the neck, and also a security hazard.

Have others found a way around this problem?

ddiakopoulos, 2016-02-26:
Absolutely should be able to run without root! Let's debug: I'm assuming you're on Ubuntu, correct? You need to ensure the udev rules are being applied correctly (or are installed). This is 95% the cause of this particular permissions issue. In short, we require udev rules to grant the correct USB permissions. 

https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules

Kukanani, 2016-02-26:
Ah, I see my problem. I was reading the installation instructions as if the "Video4Linux backend" section was an alternative option, not a required step. Everything seems to be working now. Thanks!

I am currently looking at rolling my own temporary ROS driver until yours is released...our lab is looking forward to using the software

closed 2016-02-26 by Kukanani

vacky11, 2017-04-12:
@ddiakopoulos 
I am facing the same problem. I have followed the entire installation guide inlcuding the "Video4Linux backend" section

Please help

srinimd2005, 2017-04-12:
Hi @ddiakopoulos

I am also facing the same problem with sr300 kamera.
I have installed the pre-built ros package ros-indigo-realsense

But i this error when I run roslaunch realsense_camera sr300_nodelet_default.launch

[ INFO] [1491990163.864382248]: Initializing nodelet with 8 worker threads.
[ERROR] [1491990164.005792363]: /camera/driver - No cameras detected!
[ERROR] [1491990164.005829615]: /camera/driver - Error calling rs_create_context ( api_version:11201 ): 
libusb_open(...) returned LIBUSB_ERROR_ACCESS

charlescc9, 2017-04-19:
I am also having the same issue: sr300 gives LIBUSB_ERROR_ACCESS unless I run as root. 

I have also followed the entire installation guide including the udev instructions.

Does anyone have a fix for this?

Thanks!

jcelerier, 2018-05-22:
I'm also getting this, on macOS with a D415 as of 53537d4a5115d0abbd79e5379eaa93d701344e4f