realsenseai/librealsense #45: Is it possible to run multiple SR300

sonnyhu, 2016-02-26:
I am running `librealsense` on OSX, and when I tried to run `cpp-multicam` with two SR300, I got following error:
```
RealSense error calling rs_create_context(api_version:4):
    uvc_open2(...) returned Access denied
```

Is it possible to run multiple SR300? If not, how can I modify the code so that it will fix this error and let me run SR300 sequentially (capture data sequentially)? Thanks

sonnyhu, 2016-02-26:
Just found when the second time (for the second SR300) `libusb_claim_interface` is called, it failes (`Access Denied`).

ddiakopoulos, 2016-02-27:
Thanks for the report. Multiple SR300's isn't in our validation matrix yet but we can confirm that you found a bug! We found the issue after ~10 minutes of digging -- we'll aim to push a fix to the dev branch early next week. Assume that this is a supported usage.

sonnyhu, 2016-02-27:
Awesome, thanks!! Keep me updated.

sonnyhu, 2016-03-02:
@ddiakopoulos hey, any update on this?

sgorsten, 2016-03-02:
Fixed as of cc4e3e1e53078eb60c4907430abb998acfe699f9.

closed 2016-03-02 by sgorsten

sonnyhu, 2016-03-03:
Thanks for the fix!

zyy34472, 2016-05-11:
But it does not work on F200

l3s777, 2016-08-08:
I have already followed the installation guide it worked... I turned off the pc and the next day I got this message:
RealSense error calling rs_create_context(api_version:4):
    Cannot access /sys/class/video4linux

so should I install everything again?