realsenseai/librealsense #46: Distorted pointcloud from SR300

sonnyhu, 2016-02-26:
Got color image and pointcloud from example code. The sensor was facing a corner of a wall (see color image), pointcloud is distorted. Any thoughts on calibration? Thanks

[image]
[image]

ddiakopoulos, 2016-02-27:
It sounds like you have multiple SR300s from your other filed issue. Do they all have the same problem?

sonnyhu, 2016-02-27:
Yeah, this pair of data is from another SR300 (after maximize `Motion vs Range Tradeoff` and `Laser Power` see it can get depth value further out) pointing at the same wall.

Here is pointcloud from librealsense. It is extremely bad at two edges of the wall

[image]

[image]

ddiakopoulos, 2016-02-27:
Yikes. Looks like the firmware might have shuffled some matrix numbers that we are using for projection. We will investigate more next week CC @sgorsten.

sonnyhu, 2016-02-27:
Thanks! 

I also have my own unprojection code to go from depth to point cloud which is giving good pointcloud either. But my point cloud is different from point cloud generated by librealsense. I was looking at doc of the camera. I am not familiar with distortion model. Is K (`rs_intrinsics->coeffs`) from SR300 can be used in OpenCV's `UndistortPoints` as `distCoeffs`? From what I saw, it seems K needs to be negated before used as `distCoeffs` but it is still not giving me an exact match with librealsense.

ddiakopoulos, 2016-03-10:
@sonnyhu could you print out the contents of the intrinsic struct for the depth stream?

Wollimayer, 2016-03-10:
@sonnyhu you cannot use the inverse undistorted coeffs directly in opencv. There is an implementation in rsutil.h ( rs_deproject_pixel_to_point ) for the inverse brown conrady.

I encouter the same problems with the depth date not beeing exact, both on the f200 and the new 300. It seems to have nothing to do with the intrinsic distortion but with the depth information provided by the RS.

nlyubova, 2016-03-10:
@Wollimayer I totally agree with you, after testing two different SR300 with and without rs_deproject_pixel_to_point, the result is the same, so it seems there is a problem in rs_intrinsics->coeffs

ddiakopoulos, 2016-03-10:
I agree with everyone, too! There are some parameters in the struct returned by the camera that perhaps aren't being interpreted correctly in our projection. I'll investigate more soon. https://github.com/IntelRealSense/librealsense/blob/master/src/f200-private.h#L13

sonnyhu, 2016-03-10:
@ddiakopoulos sorry, don't have it right now.
@Wollimayer Thanks! Yeah, thats what I was thinking. They are two different models.

ddiakopoulos, 2016-03-25:
@sonnyhu If you have not already, I recommend your company gets in touch with our business dev folks privately to get replacement SR300's. This is now known to be a hardware calibration issue with pre-release units. The fins on the edges will be slightly better in new units, but more importantly, flat surfaces will actually be flat.

closed 2016-03-25 by ddiakopoulos

nlyubova, 2016-03-26:
@ddiakopoulos yes that's true, the new sensor has much less distortion ! Thank you!

floe, 2017-03-21:
Sorry for digging up this old thread - @nlyubova did you use the standard click.intel.com support to get replacement cameras?

nlyubova, 2017-03-21:
Thank you, it is solved, we have got a new sensor

floe, 2017-03-21:
Yes, but did you just buy new ones or did you get replacements from Intel?