realsenseai/librealsense #47: Unity in Mac OSX

eyalfoni, 2016-02-29:
I was wondering if anyone was successfully able to get the Intel cameras working in Unity on Mac OSX?
Is there any work being done to make this possible?
Thanks!

ddiakopoulos, 2016-02-29:
At various points in time we've had our cameras working in game engines including Unity and Unreal (although usually on Windows). I am not familiar with Unity's plugin architecture on OSX (it's probably easier than the Windows DLL approach) so this should definitely be possible.

closed 2016-03-02 by ddiakopoulos

poor21, 2016-12-06:
Hello 
I was wondering is there aplugin yet to Interface Intel real sense with Unity. If yet please can you share it .
Also is it possible to do Hand tracking with lib real sense?
Awaiting your Responses
Cheers