realsenseai/librealsense #48: Have the linux kernel patches upstreamed

severin-lemaignan, 2016-03-01:
If the [kernel patch](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-camera-formats.patch) required to use the camera on Linux have been upstreamed, it would be useful to mention it in the documentation, with the name of the first kernel release including these patches.

Otherwise, this issue can serve as a tracking issue for the upstreaming of these patches (links to the possible upstream discussions welcome!)

severin-lemaignan, 2016-03-02:
For reference, the last attempt of getting the patch upstream is [discussed here](http://www.spinics.net/lists/linux-media/msg96992.html).

ddiakopoulos, 2016-03-02:
Should note that the librealsense patches also contain pixel formats for F200 and SR300, which will likely never be submitted upstream.

severin-lemaignan, 2016-03-02:
@ddiakopoulos :-( Why that? it is a pity if we can not hope to have soon enough a 'plug and play' user experience with those nice little cameras!

ddiakopoulos, 2016-03-03:
The R200 (and future generations) were intended to be compatible with Android while the F/SR series was not. Note that first discussion of the DS4 (R200) on the kernel mailing list dates back over 2 years -- this is typical for upstream patches like this. In the case of F200 and SR300, we don't have enough resources to guide yet another patch through review. Furthermore, even if those patches were accepted, newer generations of cameras would have been released in the meanwhile with potentially even more new pixel formats. So, while it's 100% preferable to have everything upstream, the reality is that it's not a priority.

closed 2016-03-03 by ddiakopoulos