realsenseai/librealsense #49: Subdevice 0 bad magic number

severin-lemaignan, 2016-03-02:
I have the following issue when trying to use a R200 with `librealsense`:

``` 
$ ./cpp-capture
Capturing DEPTH at 480 x 360, fov = 56.4 x 43.9, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.5 x 41.9, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 56.4 x 43.9, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 56.4 x 43.9, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
[...]
RealSense error calling rs_wait_for_frames(device:0x940091c):
    Timeout waiting for frames.
``` 
Note that the capture **works as expected with both the F200 and the SR300**.

Here the details of the camera (with a freshly updated firmware):
```
$ ./cpp-enumerate-devices 
Device 0 - Intel RealSense R200:
 Serial number: 2211011528
 Firmware version: 1.0.72.06
 Supported options:
[...]
```

I'm running a freshly updated Ubuntu 14.04.4 with a patched 4.4 kernel.

Any idea?

luftronix, 2016-03-02:
I have the same issue, with the same firmware version (DCM intel_rs_dcm_r200_2.1.24.6664.exe)

ddiakopoulos, 2016-03-02:
Before we debug further, do both of you have a Windows 8.1 machine to test the cameras with?

luftronix, 2016-03-02:
Yes, I do, with the latest RealSense SDK installed.

luftronix, 2016-03-02:
Similar situation with the example c-tutorial-3-pointcloud — F200 works,
while R200 timeouts.

ddiakopoulos, 2016-03-02:
So it's not clear from your note. Does R200 stream correctly in Windows using librealsense? Not the SDK.

severin-lemaignan, 2016-03-02:
I do have a windows 10 in a virtual machine. It is not yet configured for development though. I may find some time to test `librealsense` in Windows in the coming weeks, but not in the short term (as I'll be away for a conference).

ddiakopoulos, 2016-03-02:
On (rare) occasion, bad modules do make it out into the wild. It's best to verify streaming capability on a Windows 8.1 or 10 installation (**not** in a virtual machine) using either librealsense or the RealSense SDK (ideally both). `1.0.72.06` is our latest validated firmware and should correctly work across all platform backends. The delta is usually a bad module, an unsupported USB host controller, or something particular to the hardware on the machine you are using.

luftronix, 2016-03-03:
Hi Dimirti,

R200 via librealsense works well on Windows 10 with VS 2015.

The problem occurred on a machine with Ubuntu 10.04.3 LTS with kernel 4.4

I will explore further, but it is probably a bug either in uvc-v2l2 module,
or in the UVC/USB Linux stack.

Thank you,
Roman

johneleks, 2016-03-04:
The problem is gone with the last pull, most likely is was fixed in this commit

commit d62a4407756aa5d14aa07cfdd376f1e33fdc46ee
Author: Sterling Orsten <sterling.g.orsten@intel.com>
Date:   Wed Mar 2 14:18:28 2016 -0800

    Improve synchronization on R200.
    
    The R200 delivers frames matching frames consistently in Depth -> Color -> Infrared if all framerates are equal. By syncing on the "latest" stream running at the highest framerate, we can improve our frameset coherency on R200.

sclin-loauas, 2016-03-05:
I have similar problem on  newly installed Ubuntu 14.04.4, with the same firmware version 1.0.72.06.

Capturing DEPTH at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.1 x 41.5, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
RealSense error calling rs_start_device(device:0x22f2630):
    uvc_set_ctrl(...) returned LIBUSB_ERROR_PIPE

And the R200 camera is verified on Windows 10 using RealSense SDK.

Is there something wrong with LibUVC ?
@ddiakopoulos

ddiakopoulos, 2016-03-06:
@sclin-loauas could you go through the effort of installing the V4L2 backend? The libuvc backend on Linux is notoriously buggy for us (although it works rather well on OSX), and because we have a native backend with V4L2 we'd prefer to offer support on that instead.

As for @severin-lemaignan and @luftronix -- could you both pull the latest off of master and test again?

luftronix, 2016-03-07:
On Intel NUC it works almost properly: the warning shows at the beginning of streaming, once per subdevice stream. Also it works well at 90 Hz, which is exactly what we need.

It's a 4.4 kernel, patched as instructed. I did not test on unpatched kernels. 

Will test on different hardware later today.

```
john@jnuchome:~/src/librealsense$ uname -a
Linux jnuchome 4.4.0-040400-generic #201601101930 SMP Mon Jan 11 00:32:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
john@jnuchome:~/src/librealsense$  lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.4 LTS
Release:	14.04
Codename:	trusty

john@jnuchome:~/src/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-capture 
Capturing DEPTH at 480 x 360, fov = 55.0 x 42.7, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.6 x 41.9, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.0 x 42.7, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.0 x 42.7, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
john@jnuchome:~/src/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-tutorial-3-pointcloud 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense R200
    Serial number: 2211010429
    Firmware version: 1.0.72.06
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
john@jnuchome:~/src/librealsense$ 
john@jnuchome:~/src/librealsense$ 
john@jnuchome:~/src/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-capture 
Capturing DEPTH at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.6 x 41.9, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
Capturing INFRARED2 at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
john@jnuchome:~/src/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-capture 
Capturing DEPTH at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.6 x 41.9, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
Capturing INFRARED2 at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
```

luftronix, 2016-03-07:
For completeness, USB tree. I suspect, non-root hubs can add up to the issue.

```
john@jnuchome:~/src/librealsense$ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 4: Dev 6, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 6, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 6, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 6, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 6, If 4, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 6, If 5, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 1: Dev 20, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M

```

luftronix, 2016-03-07:
The results on a "ordinary" Intel box are very similar: at 60 Hz no errors, while at 90 Hz they appear. Works for me at the moment, but definitely it looks as a race condition in the software.

It also seems that kernel 4.4 is a requirement

```
john@luftbuntu:~$ uname -a
Linux luftbuntu 4.4.0+ #1 SMP Mon Mar 7 19:31:41 EET 2016 x86_64 x86_64 x86_64 GNU/Linux

john@luftbuntu:~/src/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-capture 
Capturing DEPTH at 480 x 360, fov = 55.0 x 42.7, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.6 x 41.9, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.0 x 42.7, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.0 x 42.7, distortion = NONE
Setting emitter to 0
Setting emitter to 1

# (Recompiled the example to 90 Hz)

john@luftbuntu:~/src/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-capture 
Capturing DEPTH at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.6 x 41.9, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
Capturing INFRARED2 at 320 x 240, fov = 54.3 x 42.1, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
Setting emitter to 0
Setting emitter to 1

john@luftbuntu:~/src/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-tutorial-3-pointcloud  
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense R200
    Serial number: 2211010429
    Firmware version: 1.0.72.06
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0

john@luftbuntu:~/src/librealsense$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 4, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 5, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
        |__ Port 5: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 5: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
```

Please let me know if any other information will be useful

liz-murphy, 2016-03-10:
I am also seeing this issue. I can stream color images in cpp-multicam but no depth or IR. I have the latest version of librealsense, a patched kernel and version 1.0.72.06 of the firmware.

What is the significance of rs.warn: Camera head contents version != 12, data may be missing/incorrect?

liz@c:~/codescratch/realsense_ws$ uname -a
Linux c 4.4.0-040400-generic #201601101930 SMP Mon Jan 11 00:32:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

liz@c:~/codescratch/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-capture 
rs.warn: Camera head contents version != 12, data may be missing/incorrect
Capturing DEPTH at 480 x 360, fov = 54.3 x 42.0, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.2 x 41.5, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 54.3 x 42.0, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 54.3 x 42.0, distortion = NONE

liz@c:~/codescratch/librealsense$ LD_LIBRARY_PATH=./lib ./bin/cpp-tutorial-3-pointcloud  
rs.warn: Camera head contents version != 12, data may be missing/incorrect
There are 1 connected RealSense devices.
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xe000b
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x10000e
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x14000c
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x10000d
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x250023
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x2f002d

Using device 0, an Intel RealSense R200
    Serial number: 2191300929
    Firmware version: 1.0.72.06
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0

prarobo, 2016-03-10:
I am having the same problem as @liz-murphy . I have a single realsense connected to a Ubuntu 14.04 (kernel 4.4), with most recent version of librealsense. I am only able to visualize the rbg camera stream. I don't see depth and IR streams. When I run cpp-multicam, this is the terminal output 

xxx@xxx:~/librealsense/bin$ ./cpp-multicam 
Starting Intel RealSense R200... done.
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
...

I am able to visualize the IR and depth streams using mplayer. So my realsense device is working correctly. Something should be going wrong with the driver.

I also tested my realsense on an Intel USB compute stick which gives me the same error sporadically. Sometimes it works.

I am able to get my realsense working on a different linux PC without any issues. This was repeatable and works with 4 realsenses.

Any help in solving the issue would be appreciated. Thanks.

bobdavies2000, 2016-03-10:
Make sure that the compute stick you are using has USB 3.0.  It is only in
the last month or so that they have become available with USB 3.0.  If
older, it is a USB 2.0 socket and will not work with the F200.

prarobo, 2016-03-10:
I have the USB 3.0 version of compute stick. However this problem is not limited to compute stick. I am also seeing it on an x86 64 bit Linux box.

liz-murphy, 2016-03-10:
Same, I am having issues on an i7 based desktop but no problems on an i5 laptop (both x86 64 bit Linux).

ddiakopoulos, 2016-03-10:
The warning about camera header contents version is normal since that specification changes regularly and does not impact any functionality within librealsense.

In our experience bad magic numbers come from issues with the USB 3.0 host controller or long USB cables. If the camera and software stack is working fine with one machine and completely borked on another, at best I can say there is a hardware compatibility issue and there's nothing that can be fixed on the librealsense end.

johneleks, 2016-03-10:
So far I see the error on *every* Ubuntu machine. Kernel 4.4 is better than 4.2 and better than 3.19, but still there's a race condition somewhere – wither in v4l2, uvc or librealsense itself. On Windows it works w/o these errors.

ddiakopoulos, 2016-03-10:
@johneleks what makes you think it's a race condition? Bad magic numbers are a frame integrity checking mechanism and in 99% of cases it's related to USB.

luftronix, 2016-03-10:
On exact same hardware, Ubuntu with 4.4 unpatched kernel this happens for every frame when you have both IR and color streams open. With properly patched 4.4 it happens only once, after opening. Looks as a race condition to me. To prove it I definitely have to install Windows on that machine, I am sure that error won't show up. Please let me know if I should do it.

ddiakopoulos, 2016-03-10:
@luftronix once on opening could be a symptom of how we're performing streaming sync, so I think one should be expected for every stream. It makes 100% sense that it throws tons of them when you have an unpatched kernel. Also 90hz mode isn't activated in the RealSense SDK because it *does* have some issues in regards to timing, but we feel it's better to let people experiment with it rather than fundamentally disabling. So, other than the warning message -- is anything fundamentally broken for you? 

Bad magic numbers are expected to pop up on occasion. I'm closing and locking this issue due to the number of different symptoms being documented from the 6 people who have chimed in. Please make new issues individually if you have fundamental streaming issues beyond seeing a bad magic number once in a while.

closed 2016-03-10 by ddiakopoulos