realsenseai/librealsense #50: CMake

palanglois, 2016-03-02:
Great project.

Is it possible to add CMake facilities to make integration on other projects easier ?

Best regards

ddiakopoulos, 2016-03-03:
We will likely stick to our current platform-specific project files, but if you want to create a module to help find + link librealsense for integration in other CMake projects, we would welcome that functionality as a pull request.

palanglois, 2016-03-05:
I have created a CMakeLists that uses a Findlibusb-1.0 file to compile automatically the librealsense on Windows(Visual Studio)+OS/X+Linux.

I did not take into account the compilation of the example projects to make compilation faster and to avoid the OpenGL dependency.

Tell me if you're interested in this file.

Best regards

aburbanol, 2016-03-09:
I'm interested to have It. I tried to create one, the project compiles but I got this error in execution time from QT:
> terminate called after throwing an instance of 'rs::error'
  what():  uvc_open2(...) returned Access denied

However, I run the exec with "sudo" it works. Any Idea?

best regards,

ddiakopoulos, 2016-03-09:
@aburbanol sounds like you didn't install the udev rules. Refer to the installation manual for guidance on that.

aburbanol, 2016-03-09:
@ddiakopoulos I did! I check in /etc/udev/rules.d and I also check that my user is in the group plugdev.

I did a complete CMake that includes realsense, GLFW, GLUT and OPENGL. It works pretty good with the first example in c++, but with the 2nd and the 3rd, get blocked, the only solution is to unplug and plug the camera. Ill keep doing tests. Have you ever experimented this errors?

ddiakopoulos, 2016-03-26:
@palanglois I think this is a task developers who use CMake as a build system are comfortable building individually. It usually amounts to a few dependency-finding scripts (like yours for libusb) and a 5-10 line CMake config. Closing this issue out now, but hopefully we can find someone internally who would be willing to support and maintain a formal integration.

closed 2016-03-26 by ddiakopoulos