realsenseai/librealsense #51: Connection time out during kernel patch

2Reeko, 2016-03-03:
During installation I get a time out error while trying to shallow clone the linux repo.
Specifically at this step:
>Run the following script to patch uvcvideo.ko
>./scripts/patch-uvcvideo-4.4.sh v4.4-wily (note the argument provided to this version of the script)
>This script involves shallow cloning the Linux source repository (100mb), and may take a while

I get the terminal print out saying :
> Shallow cloning Linux source repository... (~100GB, make take a while)

So I assume it is an issue in the git clone line in the script.
```
# Obtain Linux Kernel sources
echo "Shallow cloning Linux source repository... (~100GB, make take a while)"
git clone --verbose git://git.launchpad.net/~ubuntu-kernel-test/ubuntu/+source/linux/+git/mainline-crack linux-$LINUX_BRANCH --branch $THE_BRANCH --depth 1
cd linux-$LINUX_BRANCH
```

Everything up to this point worked as expected during the install. I thought there might be a newer patch than 4.4 but there doesn't seem to be. Any ideas?

Thanks,

Christian 

| Required Info  |   |
|---|---|
| Camera Model | R200| 
| Firmware Version |   | 
| Operating System & Version |  ubuntu 14.04.4  64bit |
| Kernel Version (Linux Only)  |   |
| Build System: |  VS2013, XCode or Makefile |

ddiakopoulos, 2016-03-03:
Connectivity issues can be hard to debug with git. Are you behind a firewall? For immediate next steps, try upgrading your git version through apt-get and trying again. Also try modifying the script to remove the shallow option and perhaps also the branch option.

2Reeko, 2016-03-06:
Just to update, I got around this issue. My company has a firewall that filters the git port. My git version was up to date. I modified the script from

```
git clone --verbose git://git.launchpad.net/~ubuntu-kernel-test/ubuntu/+source/linux/+git/mainline-crack linux-$LINUX_BRANCH --branch $THE_BRANCH --depth 1
```

to 

```
git clone --verbose https://git.launchpad.net/~ubuntu-kernel-test/ubuntu/+source/linux/+git/mainline-crack linux-v4.4-wily
```

The download was over 1 GiB but all seems fine now.

Thanks for the help,

Christian

closed 2016-03-06 by 2Reeko

pancx, 2016-03-15:
Have you completed it? I modified the script as you did. But after the downloading, it turned out that
`ERROR 407: Proxy Authentication Required`
I have already set the proxy, or it wouldn't to download, didn't it? Would you have some ideas about this?
thanks.

2Reeko, 2016-03-15:
I didn't want to use a proxy because of authentication issues like this. I used nmap to check if the ports used for git and http were filtered/blocked and found that my firewall allows http but filters git. I then modified the patch-uvcvideo-4.4 script as above and everything worked fine. 
Having a stab in the dark at your current problem/error message, it seems like your firewall allowed the http communication and started the download but the username and password for the proxy website you used got rejected or wasn't entered. I'd suggest trying the modified script without the proxy settings or doing the git clone at home then copying the folder into your librealsense folder at work and running the rest of the script.
Hope this helps you find a solution.