realsenseai/librealsense #54: Extracting Pointcloud Example and Having Issues

robksawyer, 2016-03-04:
| Required Info  |   |
|---|---|
| Camera Model |  F200 | 
| Firmware Version | Not sure   | 
| Operating System & Version | OSX 10.9.5  |
| Kernel Version (Linux Only)  |   |
| Build System | XCode | 

I'm running into the following issues when I try to extract the Pointcloud example and build a new project.
https://github.com/robksawyer/Realsense_Pointcloud/issues/1

I've meticulously gone through the build settings comparing the two projects and from what I can tell, I've matched them. However, still no luck. Any ideas? 

Also, I'd really like to use the camera with some Open Frameworks (openframeworks.cc) projects. Any tips to get started here? Know of any OF projects already using the camera?

ddiakopoulos, 2016-03-04:
Hi Rob. It looks like you are having basic linkage issues with GLFW (the windowing library for OpenGL). You need to add libglfw3.a to your project as an additional framework. If you installed GLFW from through homebrew, you can find the location by poking around the Xcode solution for librealsense. XCode will happily link the library for you if you drag it from a folder on disk to somewhere in your project and there's also a tab for this in the project config page.

robksawyer, 2016-03-05:
Thanks! I'll give it a shot. I'm currently working to get it compiling in Open Frameworks and have the following issue. Any ideas?

For example, this code is producing the error below.
```
        if (name) wcslcpy<sizeof(dinfo.name)/sizeof(pxcCHAR)>(dinfo.name,name);
        if (did) wcslcpy<sizeof(dinfo.did)/sizeof(pxcCHAR)>(dinfo.did,did);
```

`pxccapturemanager.h:79:26: `
```
Comparison between pointer and integer ('size_t (*)(wchar_t *, const wchar_t *, size_t)' and 'unsigned long')
```

See the project at https://github.com/robksawyer/ofxRSSDKv2 and https://github.com/robksawyer/RealsenseOFPointcloud.

robksawyer, 2016-03-05:
I was able to fix the `wcslpy` issues, but now I'm running into the following.

[image]

ddiakopoulos, 2016-03-05:
ofxRSSDK is an openFrameworks wrapper for the RealSense SDK on Windows (the production software stack for app developers). It will not compile or work on OSX as the official Intel SDK _only_ works on Windows. librealsense is a separate camera capture driver (designed specifically for OSX and Linux users). The example code you need exists as one of the projects in the samples/ directory. 

Probably what you want to do is create an openFrameworks app like normal, and then add the librealsense XCode project to your workspace. XCode will automatically figure out that the librealsense project is a dependency and link it properly. Then, you can rip out some of the librealsense sample code and insert that into your OFX app, modulo anything that has to do with GLFW, because ofx makes & handles your window and GLFW is not needed.

robksawyer, 2016-03-05:
Realizing this is a bit over my head. Thought I did it right, but apparently I don't know enough OpenGL or C++ as it's still not working and I have no clue how to remedy the latest errors. https://github.com/robksawyer/RealsenseOFPointcloud

Thanks for the help, but I'm over it.

closed 2016-03-05 by ddiakopoulos