realsenseai/librealsense #56: Unclear ubuntu install instructions

ataiya, 2016-03-05:
From [this page](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) the Ubuntu compile/install instructions say: ```all.pro contains librealsense and all example applications```. The issue is that there is **no file called all.pro** in the repository.

Compiling with Makefile is also an option, but there must be a dependency that you forgot to document:

```
ataiya@Zeus:~/dev/librealsense$ make
mkdir -p obj/libuvc
mkdir -p lib
mkdir -p bin
cc src/verify.c -std=c89 -Iinclude -c -o obj/verify.o
g++ src/context.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/context.o
g++ src/device.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/device.o
g++ src/f200.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/f200.o
g++ src/f200-private.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/f200-private.o
g++ src/image.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/image.o
g++ src/log.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/log.o
g++ src/r200.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/r200.o
g++ src/r200-private.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/r200-private.o
g++ src/rs.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/rs.o
g++ src/stream.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/stream.o
g++ src/sync.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/sync.o
g++ src/types.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/types.o
g++ src/uvc.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/uvc.o
g++ src/uvc-libuvc.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/uvc-libuvc.o
g++ src/uvc-v4l2.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/uvc-v4l2.o
In file included from src/uvc-v4l2.cpp:30:0:
/usr/include/libusb-1.0/libusb.h:732:4: warning: ISO C++ forbids zero-size array ‘dev_capability_data’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
/usr/include/libusb-1.0/libusb.h:763:4: warning: ISO C++ forbids zero-size array ‘dev_capability’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
/usr/include/libusb-1.0/libusb.h:1257:4: warning: ISO C++ forbids zero-size array ‘iso_packet_desc’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
g++ src/uvc-wmf.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/uvc-wmf.o
cc src/libuvc/ctrl.c -std=c11 -fPIC -pedantic -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/libuvc/ctrl.o
cc src/libuvc/dev.c -std=c11 -fPIC -pedantic -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/libuvc/dev.o
src/libuvc/dev.c: In function ‘uvc_open2’:
src/libuvc/dev.c:272:85: warning: ISO C99 requires rest arguments to be used [enabled by default]
       UVC_DEBUG("device has a status interrupt endpoint, but unable to read from it");
                                                                                     ^
src/libuvc/dev.c: In function ‘uvc_get_device_descriptor’:
src/libuvc/dev.c:472:7: warning: implicit declaration of function ‘strdup’ [-Wimplicit-function-declaration]
       desc_internal->serialNumber = strdup((const char*) buf);
       ^
src/libuvc/dev.c:472:35: warning: assignment makes pointer from integer without a cast [enabled by default]
       desc_internal->serialNumber = strdup((const char*) buf);
                                   ^
src/libuvc/dev.c:478:35: warning: assignment makes pointer from integer without a cast [enabled by default]
       desc_internal->manufacturer = strdup((const char*) buf);
                                   ^
src/libuvc/dev.c:484:30: warning: assignment makes pointer from integer without a cast [enabled by default]
       desc_internal->product = strdup((const char*) buf);
                              ^
src/libuvc/dev.c: In function ‘uvc_process_status_xfer’:
src/libuvc/dev.c:1496:53: warning: ISO C99 requires rest arguments to be used [enabled by default]
       UVC_DEBUG("Unhandled update from VC interface");
                                                     ^
src/libuvc/dev.c:1540:63: warning: ISO C99 requires rest arguments to be used [enabled by default]
     UVC_DEBUG("Unhandled update from VideoStreaming interface");
                                                               ^
src/libuvc/dev.c:1549:54: warning: ISO C99 requires rest arguments to be used [enabled by default]
     UVC_DEBUG("Running user-supplied status callback");
                                                      ^
cc src/libuvc/diag.c -std=c11 -fPIC -pedantic -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -c -o obj/libuvc/diag.o
src/libuvc/diag.c: In function ‘uvc_print_diag’:
src/libuvc/diag.c:186:17: warning: unknown escape sequence: '\%' [enabled by default]
                 "\t\t  GUID: ",
                 ^
[... 10 lines cut ...]
                                                          ^
src/libuvc/stream.c:877:48: warning: ISO C99 requires rest arguments to be used [enabled by default]
       UVC_DEBUG("libusb_submit_transfer failed");
                                                ^
src/libuvc/stream.c: In function ‘_uvc_populate_frame’:
src/libuvc/stream.c:946:8: warning: multi-character character constant [-Wmultichar]
   case '2YUY': /* YUY2 */
        ^
g++ -std=c++11 -shared obj/verify.o obj/context.o obj/device.o obj/f200.o obj/f200-private.o obj/image.o obj/log.o obj/r200.o obj/r200-private.o obj/rs.o obj/stream.o obj/sync.o obj/types.o obj/uvc.o obj/uvc-libuvc.o obj/uvc-v4l2.o obj/uvc-wmf.o obj/libuvc/ctrl.o obj/libuvc/dev.o obj/libuvc/diag.o obj/libuvc/frame.o obj/libuvc/init.o obj/libuvc/stream.o `pkg-config --cflags --libs libusb-1.0` -o lib/librealsense.so
cc examples/c-tutorial-1-depth.c -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/c-tutorial-1-depth
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
make: *** [bin/c-tutorial-1-depth] Error 1
ataiya@Zeus:~/dev/librealsense$
```

Which is slightly odd considering this:
```
ataiya@Zeus:~/dev/librealsense$ locate libGL.
/usr/lib/libGL.so.1
/usr/lib/x86_64-linux-gnu/libGL.la
/usr/lib/x86_64-linux-gnu/libGL.so
/usr/lib/x86_64-linux-gnu/libGL.so.1
/usr/lib/x86_64-linux-gnu/libGL.so.352.21
```

ddiakopoulos, 2016-03-05:
The QtCreator project file you are looking for is here: https://github.com/IntelRealSense/librealsense/blob/master/librealsense.qtcreator/all.pro

The issue in the Makefile also appears to have affected your attempt at installing and compiling GLFW as documented in #55 . Something about your system isn't quite right for it not to find libgl. It is a _system_ dependency created directly by Ubuntu when the OS was installed, so it is not a requirement that the installation guide direct users to install it.

A quick google reveals that this issue might be a broken symlink: http://askubuntu.com/questions/729292/i-cant-compile-a-simple-sample-no-rule-to-make-target-usr-lib-x86-64-linux

ataiya, 2016-03-09:
I updated 14.04 to the last version and re-installed the nvidia drivers.
That seem to have fixed the problem.

closed 2016-03-09 by ataiya