realsenseai/librealsense #57: IR field of view changes when depth is disabled

avivhu, 2016-03-05:
When playing around with the cpp-config-ui example, I found that the IR field of view changes when Depth is toggled on/off.

It seems to match the Depth image when Depth is on, but takes a smaller field of view when Depth is off. 

I've found the same effect with the official RealSense SDK.

Could anyone explain it? Is it a bug, or intentional?

Thank you.

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Windows 10  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013 |

teknotus, 2016-03-05:
It takes multiple IR camera pixels to calculate a depth pixel and so the
resolution of a depth image is slight less than the raw IR video. It's
intentional for the IR image to get cropped to depth resolution so that
pixels match. A feature rather than a bug.

avivhu, 2016-03-05:
The fact is that when depth is OFF, the IR field of view becomes SMALLER. That is, smaller than the IR when depth is ON, and also smaller than the Depth.
If anything, I would expect the IR FOV to become larger when depth is OFF, as I would expect it to be the raw IR size. Certainly not smaller.

ddiakopoulos, 2016-03-05:
Can you quantify 'smaller' with a number and/or pictures? Since this is happening on both the SDK and librealsense, the best I can do is pass the report to the hardware team and see if they have any thoughts. Does seem odd.

digitologist, 2016-03-05:
Is it the FOV or the focal length value that you're seeing get smaller?  If it's the focal length value then that would make sense, since it's inversely proportional to the FOV.

avivhu, 2016-03-06:
All - 
I'm attaching screenshots of the same scene with and without depth enabled.
Pay attention to the IR.
The margin on the book's right-hand side shrinks when depth is disabled, meaning the FOV shrinks.
It appears there is a different radial-distortion function running on the IR when depth is disabled.
But all I changed was to enable/disable depth and didn't touch IR settings. I would expect to get the raw IR (=distorted IR). But in any case, it should be unaffected by the depth enbled/disabled setting.

Thank you!

[image]

[image]

ddiakopoulos, 2016-03-10:
Hi @avivhu. I checked in with our camera hardware team and you found a known issue with the SR300 that boils down to a chip-level hardware limitation. From what I understand, the IR image goes through a different hardware path when depth is enabled, and all that silicon is dark when you just ask for the IR image directly. There is an open task to see if it can be fixed in firmware, but it's pretty low priority so I wouldn't count on it. Closing this issue out now, but I will post back here if a fix is ever issued.

closed 2016-03-10 by ddiakopoulos

avivhu, 2016-03-10:
ddiakopoulos - 

Many thanks for looking into it!

- Aviv