realsenseai/librealsense #58: USB 2.0 support

tommises, 2016-03-06:
I [asked ](https://software.intel.com/en-us/forums/realsense/topic/536550) about USB 2.0 compatibility on the RealSense SKD forum and was given the following answer:

> even weak USB 3.0 does not meet the requirements. USB 2.0 is not supported.

>  There is already a tremendous amount of validation required just within the given specs, pushing on other setups is outside the scope of this forum.

Is there any chance librealsense can support USB 2.0? The bandwidth is sufficient for low resolutions and frame rates and it would help integrating RealSense into embedded systems, which mostly lack USB 3.0 support.

teknotus, 2016-03-06:
> Is there any chance librealsense can support USB 2.0?

While USB 2.0 can handle the datarate at low resolutions or framerates it
cannot be done with any of the RealSense cameras. You won't even detect
them on a USB 2.0 bus. They 3.0 and up at the hardware level.

Currently that means options are limited for single board computers.

UP-board looks promising. They have a RealSense how to linked directly from
the home page and offer customization services if you need something
special for a product. $99 is one of the lowest prices for a USB 3.0
product out there.

www.up-board.org

ddiakopoulos, 2016-03-06:
Correct, our cameras are fundamentally USB 3.0 devices and do not offer any backwards compatibility. I added set of links @teknotus mentioned here because the question of small embedded devices with USB 3.0 pops up periodically.

closed 2016-03-06 by ddiakopoulos