realsenseai/librealsense #61: SR300 support

nlyubova, 2016-03-08:
could you precise the format for RGB and depth data for SR300 ? I use the following formats (as in examples) together  with the ROS wrapper from https://github.com/intel-ros/realsense, and I get weird colors and weird depth distances:
DEPTH_FORMAT = RS_FORMAT_Z16;
COLOR_FORMAT = RS_FORMAT_RGB8;
IR1_FORMAT = RS_FORMAT_Y16;

What is the correct format? 

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.10.10.0  | 
| Operating System & Version | Linux  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic |
| Build System |  VS2013, XCode or Makefile |

ddiakopoulos, 2016-03-08:
Hi. The Intel ROS node you linked doesn't support the SR300 yet (despite being based on librealsense). That ROS team will need to modify their node to deal with the correct depth scale and formats. I could possibly point you in the right direction if you send a screenshot of exactly what colors are wrong in the color or IR image. The depth scale between cameras is usually explained in other people's code by a missing multiply in the pointcloud projection code (https://github.com/IntelRealSense/librealsense/blob/master/examples/cpp-tutorial-3-pointcloud.cpp#L101). I would also recommend using one of the librealsense sample applications to double check that the camera isn't fundamentally giving weird data.

nlyubova, 2016-03-09:
Thank you, @ddiakopoulos, now it works! the problem was in pointcloud processing disabled by default. Once enabled, the values are correct!

nlyubova, 2016-03-09:
When we run examples, the RGB image looks good but in the depth data there is some distortion (like in screenshots, where a planar surface placed in front of the camera (just behind the cola) looks like a curved surface with its distortion increasing with a distance from the plane. Could you suggest how to correct this distortion? In intrinsic parameters, the Distortion matrix is empty, could you give approximate values? 

[image]
[image]

ddiakopoulos, 2016-03-09:
@nlyubova we've seen this same thing on a couple different GitHub issues now. Could you print out all members of both depth and color intrinsic structs, and the extrinsics between depth=>color?

nlyubova, 2016-03-10:
This is the camera info that I get with the ROS wrapper and librealsense (after few modifications in the wrapper's code, we can get the Distortion as well). Since we have two SR300 sensors (the same model and the same Firmware version: 3.10.10.0), here is the camera info from each of them (and there is small difference in numbers): 

**1st SR300 sensor:  Depth intrinsic: **
height: 480
width: 640
distortion_model: plumb_bob
D: [0.14829276502132416, 0.06511270254850388, 0.004560006316751242, 0.0025975550524890423, 0.08284061402082443]
K: [475.53765869140625, 0.0, 309.2130126953125, 0.0, 475.5375061035156, 245.9634552001953, 0.0, 0.0, 1.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [475.53765869140625, 0.0, 309.2130126953125, 0.0, 0.0, 475.5375061035156, 245.9634552001953, 0.0, 0.0, 0.0, 1.0, 0.0]

**1st SR300 sensor: Color intrinsic: **
height: 480
width: 640
distortion_model: plumb_bob
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [616.5604248046875, 0.0, 304.9191589355469, 0.0, 616.560546875, 241.67813110351562, 0.0, 0.0, 1.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [616.5604248046875, 0.0, 304.9191589355469, 0.0, 0.0, 616.560546875, 241.67813110351562, 0.0, 0.0, 0.0, 1.0, 0.0]

**1st SR300 sensor: depth -> color extrinsic: **
- Translation: [0.001, 0.004, -0.026]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]

**2nd SR300 sensor: Depth intrinsic: **
height: 480
width: 640
distortion_model: plumb_bob
D: [0.13900761306285858, 0.08196153491735458, 0.003836501855403185, -8.22131842141971e-05, 0.06141791120171547]
K: [475.8931884765625, 0.0, 321.28350830078125, 0.0, 475.89306640625, 245.9639434814453, 0.0, 0.0, 1.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [475.8931884765625, 0.0, 321.28350830078125, 0.0, 0.0, 475.89306640625, 245.9639434814453, 0.0, 0.0, 0.0, 1.0, 0.0]

**2nd SR300 sensor: Color intrinsic: **
height: 480
width: 640
distortion_model: plumb_bob
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [617.5955200195312, 0.0, 314.9975280761719, 0.0, 617.5956420898438, 238.17877197265625, 0.0, 0.0, 1.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [617.5955200195312, 0.0, 314.9975280761719, 0.0, 0.0, 617.5956420898438, 238.17877197265625, 0.0, 0.0, 0.0, 1.0, 0.0]

**2nd SR300 sensor: depth -> color extrinsic: **
- Translation: [0.001, 0.004, -0.025]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]

ddiakopoulos, 2016-03-17:
@nlyubova unfortunately I'm not very familiar with the ROS-ian way of performing projection. The best guidance I can offer is that you must use the projection methods in `rsutil.h` because of the SR300's somewhat confusing inverse model. 

All those coefficients look OK to me. I encountered similar pincushion distortion + warped sides in an internal validation tool used for the SR300. I have an open question to the camera hardware team to see if this is the quality of data expected. My own internal team primarily focuses on the R200+ series of devices, so we are not intimately familiar with the depth properties of the SR300.

ddiakopoulos, 2016-03-25:
Per private email communication, this is the result of known calibration issues in pre-release ("non-PRQ") SR300 modules.

closed 2016-03-25 by ddiakopoulos

Thompson104, 2017-07-13:
@nlyubova Hey, I wonder could you please tell me how to run the example code? I have qtcreator installed but I just don't know how to use it.

tispratik, 2020-03-14:
> Hi. The Intel ROS node you linked doesn't support the SR300 yet (despite being based on librealsense). That ROS team will need to modify their node to deal with the correct depth scale and formats. [...]

Updated link? https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud