realsenseai/librealsense #62: building my own application with librealsense in linux with the shared realease

aburbanol, 2016-03-09:
| Required Info  |   |
|---|---|
| Camera Model | R200| 
| Firmware Version |1.0.72.06   | 
| Operating System & Version | Ubuntu 14.04.4 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System | qmake | 

Hi I'm new with the camera, so I'm trying to make it work, but is not been simple. Even if the examples works pretty well, make my own application is been no so easy and probably I'm missing something.

For start I'm linking the shared library librealsense.so, as I wrote in the [cmake issue](https://github.com/IntelRealSense/librealsense/issues/50), I tried to integrate librealsense in other project but it always crashed. Then I tried using Qt and the qmake. I tried to add all the libs required to make it run, but it always crash. I also copy the examples source code in my file and also crashed. 

What can be useful to send you to help you to understand what is wrong?

ddiakopoulos, 2016-03-09:
So based on your other issue it looks like you're using the libuvc backend on Linux which isn't really recommended. I suspect the following is happening:

1. You ran `make` && `sudo make install` which would put realsense.so into a common directory for other applications. 
2. By default, make uses the V4L2 backend.
3. In your own apps, you are linking with `-lrealsense` and including the right files - but this is the V4L2 version installed by make
4. You haven't installed any of the prereqs for the V4L2 backend so your app is crashing

That is based on a lot of assumptions so I don't know if any of it is true for you. You might try using the makefile again by specifying that it should use the LIBUVC backend (`make BACKEND=LIBUVC` or something like that) and then `sudo make install`

aburbanol, 2016-03-10:
@ddiakopoulos  I'm using the V4L2 backend.

DongDongJu, 2016-03-10:
@aburbanol  i'd same error.

r u fix that?

DongDongJu, 2016-03-11:
I'm new with this camera module too.
I read this repo's install introduction for linux.
but same error about LIBUSB_ERROR_PIPE.
plz give me some information about this error?

best regards.
@ddiakopoulos

aburbanol, 2016-03-11:
@DongDongJu No I haven't! still trying to make it work. @ddiakopoulos any other think I can do?

aburbanol, 2016-03-11:
Dear @ddiakopoulos, @DongDongJu 

As I told you before it compiles, So I decided just to ask for color and depth and it is working but only for one time after that gave me the same errors first `rs::error was thrown when calling rs_wait_for_frames(device:0x1f40bf0)` and then `uvc_set_ctrl(...) returned LIBUSB_ERROR_PIPE` systematically, then I have to unplug and plug and it works again.

 So the problem is when I ask several streams at the same time and by experience usually cameras libraries has a way to terminate safely the connection, like a close function or something. @ddiakopoulos  there is any reason why you dont use something like that to safely terminate the connection with the camera?

ddiakopoulos, 2016-03-11:
@aburbanol you encountered the reason why we don't recommend people use the libuvc backend on Linux -- it doesn't always terminate the camera correctly. Using the C++ API, the device is released on object destruction. Unfortunately with libuvc the device handle doesn't always get released so you need to unplug/replug -- known issue. The only recommended workaround is to use the V4L2 backend which you were having issues with previously.

Occasional bad numbers are expected, but regularly seeing them probably points to an issue with your USB connection or hardware platform (port, cable, host controller) and unfortunately not something we can solve in software.

@DongDongJu please open a new issue for your specific problem with all relevant details about your hardware, platform, and debugging steps so far. The same error messages are often used for a **large** variety of unrelated issues.

aburbanol, 2016-03-14:
@ddiakopoulos But then I dont Understand how I end up using libuvc if I followed the steps for using V4L2. I know I missing something, but where? I manage to make the camera work only using the Depth and the IR streams. but the unplugging problem keeps the same. I guess is because that I'm using livUVC but how come?

ddiakopoulos, 2016-03-15:
@aburbanol I'm lost. Can you verify the accuracy of my summary: 
  * With the `cpp-capture` program, you can correctly stream color, depth, and IR
  * You get the error `rs::error was thrown when calling rs_wait_for_frames` after a while (1 minute?)
  * On the next launch of the app, you get `LIBUSB_ERROR_PIPE` and cannot open the camera. 
  * The way you fix this is by unplugging the camera and plugging it back in

If the following is all true, then you successfully installed the V4L2 backend (forget anything I said previously about the libuvc/libusb backend). The problem might be hardware related at that point.

aburbanol, 2016-03-16:
@ddiakopoulos As I got confused about the LivUVC thing, I installed it. and It was not working. now I delete the uvc.conf and recompile without the backend parameter and is everything working! 

I Leave you the CmakeLists.text if you wanna add it in the tutorial or as a resource. 

````
project(IntelModeTest)
cmake_minimum_required(VERSION 2.8)
add_compile_options(-std=c++11 -fPIC -lusb-1.0 -lpthread )

#########################################################
# FIND Threads libusb-1.0
#########################################################
find_package(Threads REQUIRED)

find_package(PkgConfig)
pkg_check_modules(PC_LIBUSB REQUIRED libusb-1.0)
include_directories( ${LIBUSB_1_INCLUDE_DIRS})

#########################################################
# FIND OPENGL
#########################################################
find_package(OpenGL REQUIRED)
include_directories(${OpenGL_INCLUDE_DIRS})
link_directories(${OpenGL_LIBRARY_DIRS})
add_definitions(${OpenGL_DEFINITIONS})
if(NOT OPENGL_FOUND)
    message(ERROR " OPENGL not found!")
endif(NOT OPENGL_FOUND)
#########################################################
# FIND GLUT
#########################################################
find_package(GLUT REQUIRED)
include_directories(${GLUT_INCLUDE_DIRS} )
#########################################################
# FIND GLFW
#########################################################
find_package(PkgConfig REQUIRED)
 pkg_search_module(GLFW REQUIRED glfw3)
include_directories(${GLFW_INCLUDE_DIRS})
#########################################################
# FIND OpenCV
#########################################################
find_package( OpenCV REQUIRED )
#########################################################
# FIND realsense
#########################################################
include_directories("/home/aburbanol/librealsense/include")
link_directories("/usr/local/lib")
set(LIBS ${LIBS} realsense)

link_directories( ${realsense_LIBRARY})
include_directories( ${realsense_INCLUDE_DIR})

aux_source_directory(. SRC_LIST)
add_executable(${PROJECT_NAME} ${SRC_LIST})
target_link_libraries( ${PROJECT_NAME}
${LIBS} ${OpenCV_LIBS} ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${GLFW_LIBRARIES} ${CMAKE_THREAD_LIBS_INIT} )
```
`

closed 2016-03-16 by aburbanol