realsenseai/librealsense #63: Cannot display depth image

SandW, 2016-03-11:
| Required Info  |  cannot display depth image |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.06 | 
| Operating System & Version |  Ubuntu14.04 .4|
| Kernel Version (Linux Only)  |  4.4 |
| Build System |  Makefile | 

output msg of `dmesg | tail -n 50`

> [ 5468.500339] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 2: -32 (exp. 4).
[ 5468.550888] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 2: -32 (exp. 4).
[ 5468.601484] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 2: -32 (exp. 4).
[ 5468.652028] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 2: -32 (exp. 4).
[ 5468.702570] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 2: -32 (exp. 4).
[ 5470.849161] uvcvideo: Non-zero status (-71) in video completion handler.
[ 5470.861121] uvcvideo: Non-zero status (-71) in video completion handler.
[ 5472.096904] nouveau 0000:01:00.0: DRM: resuming kernel object tree...
[ 5472.449577] nouveau 0000:01:00.0: DRM: resuming client object trees...
[ 5472.449687] nouveau 0000:01:00.0: DRM: resuming display...
[ 5472.449709] nouveau 0000:01:00.0: DRM: resuming console...
[ 5478.197742] uvcvideo: Non-zero status (-71) in video completion handler.
[ 5478.296198] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires \[Package\] (20150930/nsarguments-95)
[ 5478.296298] ACPI: \_SB_.PCI0.PEG0.PEGP: failed to evaluate _DSM
[ 5478.296302] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires \[Package\] (20150930/nsarguments-95)
[ 5478.296445] nouveau 0000:01:00.0: DRM: suspending console...
[ 5478.296449] nouveau 0000:01:00.0: DRM: suspending display...
[ 5478.296459] nouveau 0000:01:00.0: DRM: evicting buffers...
[ 5478.296461] nouveau 0000:01:00.0: DRM: waiting for kernel channels to go idle...
[ 5478.296485] nouveau 0000:01:00.0: DRM: suspending client object trees...
[ 5478.302092] nouveau 0000:01:00.0: DRM: suspending kernel object tree...
[ 5480.588100] nouveau 0000:01:00.0: DRM: resuming kernel object tree...
[ 5480.940919] nouveau 0000:01:00.0: DRM: resuming client object trees...
[ 5480.941007] nouveau 0000:01:00.0: DRM: resuming display...
[ 5480.941028] nouveau 0000:01:00.0: DRM: resuming console...
[ 5484.687961] uvcvideo: Non-zero status (-71) in video completion handler.
[ 5484.699959] uvcvideo: Non-zero status (-71) in video completion handler.
[ 5487.259403] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires \[Package\] (20150930/nsarguments-95)
[ 5487.259517] ACPI: \_SB_.PCI0.PEG0.PEGP: failed to evaluate _DSM
[ 5487.259522] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires \[Package\] (20150930/nsarguments-95)
[ 5487.259671] nouveau 0000:01:00.0: DRM: suspending console...
[ 5487.259674] nouveau 0000:01:00.0: DRM: suspending display...
[ 5487.259684] nouveau 0000:01:00.0: DRM: evicting buffers...
[ 5487.259686] nouveau 0000:01:00.0: DRM: waiting for kernel channels to go idle...
[ 5487.259712] nouveau 0000:01:00.0: DRM: suspending client object trees...
[ 5487.265403] nouveau 0000:01:00.0: DRM: suspending kernel object tree...
[ 5691.281852] nouveau 0000:01:00.0: DRM: resuming kernel object tree...
[ 5691.630730] nouveau 0000:01:00.0: DRM: resuming client object trees...
[ 5691.630839] nouveau 0000:01:00.0: DRM: resuming display...
[ 5691.630870] nouveau 0000:01:00.0: DRM: resuming console...
[ 5693.570589] systemd-hostnamed[6625]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[ 5697.333227] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires \[Package\] (20150930/nsarguments-95)
[ 5697.333374] ACPI: \_SB_.PCI0.PEG0.PEGP: failed to evaluate _DSM
[ 5697.333380] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires \[Package\] (20150930/nsarguments-95)
[ 5697.333593] nouveau 0000:01:00.0: DRM: suspending console...
[ 5697.333598] nouveau 0000:01:00.0: DRM: suspending display...
[ 5697.333611] nouveau 0000:01:00.0: DRM: evicting buffers...
[ 5697.333613] nouveau 0000:01:00.0: DRM: waiting for kernel channels to go idle...
[ 5697.333639] nouveau 0000:01:00.0: DRM: suspending client object trees...
[ 5697.339375] nouveau 0000:01:00.0: DRM: suspending kernel object tree...

After press the button of "start capture" in "cpp-config-ui", no depth image displayed, and it crashed after changing some parameters named with "R200_".

Depth image cannot be displayed in every app.

[image]

When I running the "cpp-alignimages", messages constantly showed in the terminal:
> rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xed00e6
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xf300ee
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xfa00f7

ddiakopoulos, 2016-03-11:
Do you have a Windows machine handy to double check that the camera is working? Because the camera has so many additional steps on Ubuntu sometimes it's better to quickly sanity check that it's not the camera hardware (which happens on occasion... could also be USB cable, USB host controller, etc).

SandW, 2016-03-12:
I can use the camera under Win10 using intel_rs_dcm_r200_2.1.24.6664 and intel_rs_sdk_runtime_8.0.24.6528. It works pretty well.

ddiakopoulos, 2016-03-17:
@SandW maybe try running the `./scripts/patch-uvcvideo-4.4.sh v4.4-wily` script again. Looks like that step might have failed without providing an obvious error.

The crashes from cpp-config-ui, while confusing to new users, are usually the result of altering R200_ controls that were only meant to be modified before or after streaming has started. Our example UI isn't smart enough to disable controls according to when they are available to set.

SandW, 2016-03-17:
Thanks!
I have followed your instructions. However, no image was shown in the cpp-config-ui after click start capture. It exited automatically after a few seconds. Then no device can be found. I plugged in the USB again and no device can be found, unless I reboot my PC.

> (Before plug in the R200 and after the reinstallation)
[   18.585583] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[   18.585584] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[   18.773545] init: Failed to obtain startpar-bridge instance: Unknown parameter: INSTANCE
[   20.970631] init: cups main process (838) killed by HUP signal
[   20.970641] init: cups main process ended, respawning
[   23.246585] Bluetooth: Core ver 2.21
[   23.246610] NET: Registered protocol family 31
[   23.246612] Bluetooth: HCI device and connection manager initialized
[   23.246617] Bluetooth: HCI socket layer initialized
[   23.246620] Bluetooth: L2CAP socket layer initialized
[   23.246628] Bluetooth: SCO socket layer initialized
[   23.376344] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   23.376348] Bluetooth: BNEP filters: protocol multicast
[   23.376352] Bluetooth: BNEP socket layer initialized
[   23.445132] Bluetooth: RFCOMM TTY layer initialized
[   23.445138] Bluetooth: RFCOMM socket layer initialized
[   23.445144] Bluetooth: RFCOMM ver 1.11
[   26.041531] r8169 0000:03:00.0 eth1: link down
[   26.041586] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   26.927233] MAC Address = 48:8a:d2:08:d3:5d
[   26.931273] IPv6: ADDRCONF(NETDEV_UP): wlan3: link is not ready
[   26.938404] R8188EU: ERROR indicate disassoc
[   26.939179] IPv6: ADDRCONF(NETDEV_UP): wlan3: link is not ready
[   28.946347] R8188EU: ERROR assoc success
[   28.946376] IPv6: ADDRCONF(NETDEV_CHANGE): wlan3: link becomes ready
[   29.390529] audit_printk_skb: 15 callbacks suppressed
[   29.390534] audit: type=1400 audit(1458254470.303:17): apparmor="STATUS" operation="profile_replace" name="/sbin/dhclient" pid=988 comm="apparmor_parser"
[   29.390541] audit: type=1400 audit(1458254470.303:18): apparmor="STATUS" operation="profile_replace" name="/usr/lib/NetworkManager/nm-dhcp-client.action" pid=988 comm="apparmor_parser"
[   29.390546] audit: type=1400 audit(1458254470.303:19): apparmor="STATUS" operation="profile_replace" name="/usr/lib/connman/scripts/dhclient-script" pid=988 comm="apparmor_parser"
[   29.390875] audit: type=1400 audit(1458254470.303:20): apparmor="STATUS" operation="profile_replace" name="/usr/lib/NetworkManager/nm-dhcp-client.action" pid=988 comm="apparmor_parser"
[   29.390880] audit: type=1400 audit(1458254470.303:21): apparmor="STATUS" operation="profile_replace" name="/usr/lib/connman/scripts/dhclient-script" pid=988 comm="apparmor_parser"
[   29.391040] audit: type=1400 audit(1458254470.303:22): apparmor="STATUS" operation="profile_replace" name="/usr/lib/connman/scripts/dhclient-script" pid=988 comm="apparmor_parser"
[   29.391799] audit: type=1400 audit(1458254470.303:23): apparmor="STATUS" operation="profile_replace" name="/usr/lib/cups/backend/cups-pdf" pid=993 comm="apparmor_parser"
[   29.391806] audit: type=1400 audit(1458254470.303:24): apparmor="STATUS" operation="profile_replace" name="/usr/sbin/cupsd" pid=993 comm="apparmor_parser"
[   29.392139] audit: type=1400 audit(1458254470.303:25): apparmor="STATUS" operation="profile_replace" name="/usr/sbin/cupsd" pid=993 comm="apparmor_parser"
[   30.004495] audit: type=1400 audit(1458254470.915:26): apparmor="STATUS" operation="profile_load" name="/usr/sbin/tcpdump" pid=995 comm="apparmor_parser"
[   34.053290] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   34.102456] NFSD: starting 90-second grace period (net ffffffff81ef05c0)
[   38.119886] init: plymouth-upstart-bridge main process ended, respawning
[   38.128717] init: plymouth-upstart-bridge main process (1454) killed by TERM signal
[   50.997289] audit_printk_skb: 123 callbacks suppressed
[   50.997293] audit: type=1400 audit(1458254491.911:68): apparmor="STATUS" operation="profile_replace" name="/usr/lib/cups/backend/cups-pdf" pid=2058 comm="apparmor_parser"
[   50.997298] audit: type=1400 audit(1458254491.911:69): apparmor="STATUS" operation="profile_replace" name="/usr/sbin/cupsd" pid=2058 comm="apparmor_parser"
[   50.997802] audit: type=1400 audit(1458254491.911:70): apparmor="STATUS" operation="profile_replace" name="/usr/sbin/cupsd" pid=2058 comm="apparmor_parser"
[  134.034979] systemd-hostnamed[2601]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[  812.094184] media: Linux media interface: v0.10
[  812.132627] Linux video capture interface: v2.00
[  812.176129] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel
[  812.176745] usbcore: registered new interface driver uvcvideo
[  812.176747] USB Video Class driver (1.1.1)

(After plug in R200)
[   23.376352] Bluetooth: BNEP socket layer initialized
[   23.445132] Bluetooth: RFCOMM TTY layer initialized
[   23.445138] Bluetooth: RFCOMM socket layer initialized
[   23.445144] Bluetooth: RFCOMM ver 1.11
[   26.041531] r8169 0000:03:00.0 eth1: link down
[   26.041586] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   26.927233] MAC Address = 48:8a:d2:08:d3:5d
[   26.931273] IPv6: ADDRCONF(NETDEV_UP): wlan3: link is not ready
[... cut ...]
[ 1066.032906] uvcvideo: Unable to create debugfs 11-2 directory.
[ 1066.033143] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:07.0/0000:06:00.0/usb11/11-2/11-2:1.4/input/input23

$ ./cpp-config-ui 
rs.error: VIDIOC_REQBUFS error 19, No such device
rs.error: VIDIOC_REQBUFS error 19, No such device
RealSense error calling rs_wait_for_frames(device:0x832a00):
    Timeout waiting for frames.

$ ./cpp-alignimages 
No device detected. Is it plugged in?

$ ./cpp-config-ui 
No device found. Is it plugged in?

ddiakopoulos, 2016-03-17:
@SandW this sounds suspiciously like you're on hardware that is incompatible with the camera. I've never seen a case where a reboot is necessary to re-enumerate one of the cameras.

SandW, 2016-03-18:
Have you tested in in NUC under Ubuntu 14.04? I can try my NUC.

bobdavies2000, 2016-03-18:
My NUC running Ubuntu 14.04 works fine with the r200 and the f200.

ddiakopoulos, 2016-04-14:
@SandW we've found in a couple cases that sometimes these issues can be fixed by twiddling with BIOS settings like "USB Legacy Support" See this issue: #105

SandW, 2016-04-18:
Thanks! I will try.

closed 2016-05-20 by ddiakopoulos