realsenseai/librealsense #65: Weird device ids

vidstige, 2016-03-11:
Using Xubuntu with 4.4 kernel.

I'm seeing some weird device ids coming out of librealsense containing lots of "FF", also on one camera (SR300) there is a space character at the end of the serial number.

Querying the serial number directly with udevadm like so yields more sane serial numbers.

    udevadm info --query=all --name=/dev/video2 | grep -oP 'ID_SERIAL_SHORT=\K([0-9]+)'

| Source | Serial number |
|---|---|
| udevadm | 039140052101 |
| librealsense | 03FFFFFF9140052101 |

What gives? Which is correct? Can you please avoid whitespace in serial numbers? Please advice.

vidstige, 2016-03-11:
I found the problem and made a pull request - #66

However, for SR300 cameras serial numbers can still contain a trailing space. This occurs when byte 5 of the aw serial numbers is 0-15, because of the weird format string, see below.

    sprintf(formattedBuffer, "%02X%02X%02X%02X%02X%-2X", ss[0], ss[1], ss[2], ss[3], ss[4], ss[5]);

I don't know what the intention was here? How should a serial number ending with a byte 0x02 look? Like this: `aaaaaaaaaa-02`, or like this `aaaaaaaaaa2`? Right now they look like this `aaaaaaaaaa2 ` (please note trailing space).

vidstige, 2016-03-11:
The former makes more sense as that would allow one to distinguish between SR300 and F200 serial numbers.

ddiakopoulos, 2016-03-11:
@vidstige We have no control over the serial numbers that the firmware (or usb descriptor) hands back so we'll just need to work through this in software. I'm inclined to trim the trailing space on SR300 numbers. We can already distinguish between F200 and SR300 based on VID/PID.

vidstige, 2016-03-11:
@ddiakopoulos thanks for your answer. Did you check out the issue description and related pull request? This problem is not on the camera firmware, but rather it is inside the librealsense software. I've even identified the problematic line.

vidstige, 2016-03-11:
I would not see this as a enhancement, but rather a quite serious bug.

vidstige, 2016-03-11:
Checking through the history real quick it seems you added this code last year. Look at36b467ed74f220f30ae181dfb59256d8ccd245b4

ddiakopoulos, 2016-03-11:
Yep, I'll merge when we're ready. I wouldn't say it's "serious bug", just a minor formatting issue. We'll fix it up! Closing this out now because you already have a pull request in flight to fix it.

closed 2016-03-11 by ddiakopoulos

vidstige, 2016-03-11:
:+1: thanks for looking into this! :)

vidstige, 2016-03-12:
Oh, will you also fix the SR300 triailing whitespace thingy?

ddiakopoulos, 2016-03-15:
@vidstige yep, soon.

vidstige, 2016-03-21:
Thanks for merging the FFFFFF fix, but there is still the problem with SR300 returning serial numbers with trailing whitespace, which is not so nice. Can you please also fix that? Or can you show an example of how the serial numbers should look like, and I'll try to make a minimal pull request for that.

taketwo, 2017-02-09:
Any follow-up on the trailing whitespace? I still get it the latest version.