realsenseai/librealsense #67: error for R200, Timeout waiting for frames.

sclin-loauas, 2016-03-11:
I was trying to use a R200 camera with firmware version 1.0.72.06.
My system is ubuntu 14.04 with a patched 4.4 kernel.

After I run cpp-capture. A graphic window comes out and freezes in the first frame. 
The program crashes after a few seconds with the following outputs on terminal:

Capturing DEPTH at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.1 x 41.5, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
RealSense error calling rs_wait_for_frames(device:0x8351f0):
    Timeout waiting for frames.

ddiakopoulos, 2016-03-11:
Timeout waiting for frames is most often a symptom of an unsupported USB 3.0 host controller. We have also encountered plenty of systems where only 2 out of the 4 available physical USB 3.0 ports on a given machine will work! So, in order of priority:

  1. Only stream depth (remove extra bandwidth)
  2. Try a different (possibly thicker or shorter) USB 3.0 cable. Skip this step if you are using the one that came bundled with the R200.
  3. Try all the different 3.0 ports available to you
  4. Attempt to use a powered USB 3.0 hub (sometimes controllers don't negotiate enough power to the camera to actually stream data).
  5. Hop over to a Windows 8/10 machine to double check that the camera can stream at all

bpwiselybabu, 2016-03-14:
I have the same problem.

I had a working R200 in ubuntu 14.04 on my workstation. i wanted to test it on laptop, so I performed the same process on it and R200 never worked in it. I gave up on the laptop and returned to my workstation and all of a sudden the R200 now does not work. The F200 works without any issue on the same computer. Is there anyway to reflash the firmware?

ddiakopoulos, 2016-03-15:
@bpwiselybabu 'suddently does not work' needs some more clarification! What errors are you encountering?

This library does not write any permanent data back to the camera so it's unlikely the firmware went corrupt. You can always plug it into a Windows 10 machine and install the Intel RealSense DCM to reflash the firmware.

sclin-loauas, 2016-03-16:
@ddiakopoulos 
I tried a powered usb 3.0 port. The program works well if only depth or only color stream is enabled, but still crashes with all streams enabled. It seems that the usb port on my laptop is not fast enough.
I want to change the framerate, but I didnot find the config interface from cpp-config-ui or corresponding rs-option in file "rs.h". Would you help me with that?

bpwiselybabu, 2016-03-16:
@ddiakopoulos I am sorry, I was not very clear about my problem as I was working for a conference deadline. I resolved the situation using some of the old driver that was published in the RealSense_ROS package but I will explain the issue I faced and the way I resolved it a bit more in detail.

1. When I encountered this issue, I had a working R200 on my workstation with no issues. The workstation runs Linux 14.04 with 3.19.0-49-generic kernel.  
2. I wanted to get the R200 that I have working on a Linux laptop(portability duh). So I updated the firmware on the R200 using a windows machine and then progressively updated the kernel on my linux laptop from trusty kernel(3.16) to 3.18 and then to whily-lts(4.2) and then to custom kernel 4.4.4. During each iteration, i followed all the instructions presented in this respo for installation with no success. At some point I was able to get the RGB camera working but not the depth camera but later ended up in a point where it would give a timeout error. 
3. I gave up on my laptop and returned to my workstation and to my surprise, the R300 stopped working on my workstation on which I have not done any significant changes. F200 continued to work without any issues. (This is where I suspected a possible corruption to the firmware as I had attempted multiple installs and had encountered multiple timeouts with different sync errors). I decided to reinstall the driver on my workstation and change the backend to UVC from V4L with no success.
4. Finally based on my memory about how I had installed the realsense first time. I found the install script presented in https://github.com/PercATI/RealSense_ROS/tree/master/r200_install and recompile the driver with V4L backend. This resolved my issue on the workstation.

Wollimayer, 2016-03-16:
The cake is a Lie! No, wait a minute... This is not a cake, this is an e-mail account. 

The e-mail account is a lie.

Im sorry.

bpwiselybabu, 2016-03-16:
It might seem like a lie but I am pretty certain there is some issue in your install instructions that could have stopped R200 from working. If you are not writing back to R200. It means the drivers on my workstation got corrupted. And the only possibility of it is repeated reinstall of your drivers.

sclin-loauas, 2016-03-17:
I change the fps to 30, then everything if fine. 
It seems that R200 is strict with speed of USB port.
But I still don't understand why it is OK to stream at 60fps on a Windows laptop.
Thank you for your help! @ddiakopoulos

ddiakopoulos, 2016-03-17:
@bpwiselybabu Our installer script is a modified version of the one you linked in the RealSense_ROS repo (just take a look at the source). I think this prompts a need to introduce an installation script so we can reset the state of any possibly failed or incomplete installation sub-steps. As you noted, backing out entirely and just trying it all again often fixes any possible mis-steps along the installation process. 

@sclin-loauas Yes, R200 will happily time out if your USB controller can't handle the bandwidth. The 60/30 behavior is odd to me, but there's many variables which could make an impact. 

Closing this out.

closed 2016-03-17 by ddiakopoulos

gabrielarpino, 2016-07-20:
Hi @sclin-loauas , how do you change the fps to 30? I am having the same problem.

dianchen96, 2016-07-28:
@sclin-loauas Could you please detail on how to change the fps? I am experiencing the same issue. Thanks in advance!

CWallbridge, 2016-12-07:
I had the same problem with an SR300 camera on a newly built computer. SR300-live-test reported "Timeout waiting for frames" on all of the colour image tests. Problem was solved with a powered USB 3.0 hub.

yangman111, 2017-12-04:
I had the same problem with an SR300 camera on a newly built computer. SR300-live-test reported "Timeout waiting for frames" on all of the colour image tests. Problem was solved with a powered USB 3.0 hub.