realsenseai/librealsense #70: Depth stream not working

prarobo, 2016-03-14:
Camera Model: R200
Firmware Version: 1.0.71.06
Operating System & Version: Linux 14.04.04 x86_64
Kernel Version (Linux Only): 4.4

I am only able to visualize the rbg camera stream. I don't see depth and IR streams. When I run cpp-multicam, this is the terminal output

xxx@xxx:~/librealsense/bin$ ./cpp-multicam
Starting Intel RealSense R200... done.
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
...
[this warning keeps repeating]

I am able to visualize the IR and depth streams using mplayer. So my realsense device is working correctly. Something should be going wrong with the driver. 

Is it a syncing issue that makes the driver drop frames?

ddiakopoulos, 2016-03-14:
Hi, I strongly recommend using a Windows machine to update your firmware. 1.0.72.06 is the latest and 1.0.71.xx has known issues. Installing the [DCM](https://downloadcenter.intel.com/download/25044/Intel-RealSense-Depth-Camera-Manager-DCM-) on a Windows 10 machine with a camera plugged in will update the firmware. If you don't have Win10 handy, I can dig up the utility for Win 8.1 (seems like they recently removed it!).

ddiakopoulos, 2016-03-14:
@prarobo so I accidentally replicated this issue today installing a new machine and saw this exact behavior (RGB only, bad magic numbers for all streams) and it was because I forgot to run the 4.4 uvcvideo patching script: 

`./scripts/patch-uvcvideo-4.4.sh v4.4-wily`

prarobo, 2016-03-14:
@ddiakopoulos Thanks for your response. I updated the firmware to 1.0.72.06. I also reapplied the uvcvideo patch. I am still seeing the same error. Bad magic number and no depth stream.

Any thoughts? I am able to see the individual streams (color, IR and depth through mplayer) but cpp-multicam shows only color stream.

ddiakopoulos, 2016-03-14:
@prarobo what about cpp-capture with a single camera?

prarobo, 2016-03-14:
Only color image shows, no depth image when running cpp-capture. This is the terminal output.

Capturing DEPTH at 480 x 360, fov = 55.1 x 42.8, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.5 x 41.8, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.1 x 42.8, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.1 x 42.8, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x20001c
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xf000e
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x12000d
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x150012
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xd000d
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x110012
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xe0013
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x130013
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x120010
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xf000e
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x11000e
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x110011
......
[repeats]

ddiakopoulos, 2016-03-15:
This is the exact behavior of a system that hasn't been correctly patched. I'm not sure the script is successfully being applied on your system. Can you run the following commands and post your dmesg output?

`sudo modprobe -r uvcvideo`
`dmesg` (only copy relevant lines here, should notate that you removed the kernel object)
`sudo modprobe uvcvideo`
`dmesg` (should let you know that it re-inserted correctly)

Also after streaming, could you most the last few lines of dmesg? Unknown frame formats are also a telltale sign.

prarobo, 2016-03-16:
This what I am seeing from dmesg (see below). I am seeing an unknown video format. I am applying the patch the though. Is there a way to check if the patch is getting applied?

[  368.323530] usb 4-3.3: new SuperSpeed USB device number 5 using xhci_hcd
[  368.342214] usb 4-3.3: New USB device found, idVendor=8086, idProduct=0a80
[  368.342217] usb 4-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  368.342218] usb 4-3.3: Product: Intel RealSense 3D Camera R200
[  368.342220] usb 4-3.3: Manufacturer: Intel Corp
[  368.342221] usb 4-3.3: SerialNumber: SN_2191301595
[  368.344361] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[  368.344363] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[  368.344368] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  368.347695] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[  368.347699] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  368.348934] uvcvideo: Unable to create debugfs 4-5 directory.
[  368.349007] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.3/4-3.3:1.2/input/input40
[  368.351038] uvcvideo: Unknown video format 30315752-0000-0010-8000-00aa00389b71
[  368.351041] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  368.352331] uvcvideo: Unable to create debugfs 4-5 directory.
[  368.352401] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.3/4-3.3:1.4/input/input41

ddiakopoulos, 2016-03-16:
You need to monitor what is failing in the patch script. The patch is not being applied correctly. When the script is finished, it should copy a new version of uvcvideo.ko to your system directory and then you manually re-load it with the `sudo modprobe uvcvideo` command. The output above is definitely from an unpatched version which explains all of your issues.

javcursor, 2016-03-19:
Hello, I have the same problem, executing cpp-capture I only see rgb image:
vigus@NUC-VIGUS1:~/librealsense-master/bin$ ./cpp-capture
Capturing DEPTH at 480 x 360, fov = 55.5 x 43.1, distortion = NONE
Capturing COLOR at 640 x 480, fov = 53.8 x 41.3, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.5 x 43.1, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.5 x 43.1, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0xf0006
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0xf0014
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x130015
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x150013
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x26001e
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x400034
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x3d0035
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x3f0036
.
.
.

I have applied several times the patch uvcvideo-4.4.sh v4.4-wily and I haven't resolved the issue. The dmesg info:
vigus@NUC-VIGUS1:~/librealsense-master/bin$ sudo dmesg | tail -n 50
[ 1235.233618] usb 3-1: Product: Intel RealSense 3D Camera R200
[ 1235.233620] usb 3-1: Manufacturer: Intel Corp
[ 1235.233622] usb 3-1: SerialNumber: SN_2431013021
[ 1235.235958] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 1235.235961] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 1235.235964] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1235.239232] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[ 1235.239235] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1235.240600] uvcvideo: Unable to create debugfs 3-3 directory.
[ 1235.240673] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2/input/input19
[ 1235.242552] uvcvideo: Unknown video format 30315752-0000-0010-8000-00aa00389b71
[ 1235.242554] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1235.243971] uvcvideo: Unable to create debugfs 3-3 directory.
[ 1235.244042] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.4/input/input20
[ 1323.356012] uvcvideo: Non-zero status (-71) in video completion handler.
[ 1323.368013] uvcvideo: Non-zero status (-71) in video completion handler.
[ 1724.867882] usbcore: deregistering interface driver uvcvideo
[ 1748.602913] media: Linux media interface: v0.10
[ 1748.606820] Linux video capture interface: v2.00
[ 1748.699088] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 1748.699091] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 1748.699095] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1748.701889] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[ 1748.701892] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1748.703314] uvcvideo: Unable to create debugfs 3-3 directory.
[ 1748.703446] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2/input/input21
[ 1748.704778] uvcvideo: Unknown video format 30315752-0000-0010-8000-00aa00389b71
[ 1748.704782] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1748.706203] uvcvideo: Unable to create debugfs 3-3 directory.
[ 1748.706262] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.4/input/input22
[ 1748.707193] usbcore: registered new interface driver uvcvideo
[ 1748.707195] USB Video Class driver (1.1.1)
[ 1808.188787] usb 3-1: USB disconnect, device number 3
[ 1817.500891] usb 3-2: new SuperSpeed USB device number 4 using xhci_hcd
[ 1817.519540] usb 3-2: New USB device found, idVendor=8086, idProduct=0a80
[ 1817.519544] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1817.519546] usb 3-2: Product: Intel RealSense 3D Camera R200
[ 1817.519548] usb 3-2: Manufacturer: Intel Corp
[ 1817.519550] usb 3-2: SerialNumber: SN_2431013021
[ 1817.521902] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 1817.521906] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 1817.521911] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1817.525353] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[... 339 lines cut ...]
patching file debian/stamps/keep-dir
patching file debian/tests-build/README
patching file debian/tests-build/check-aliases
patching file debian/tests/control
patching file debian/tests/rebuild
patching file debian/tests/ubuntu-regression-suite
patching file debian/tools/generic
patching file debian.master/changelog
patching file debian.master/config/amd64/config.flavour.generic
patching file debian.master/config/amd64/config.flavour.lowlatency
patching file debian.master/config/arm64/config.common.arm64
patching file debian.master/config/armhf/config.common.armhf
patching file debian.master/config/config.common.ubuntu
patching file debian.master/config/i386/config.flavour.generic
patching file debian.master/config/i386/config.flavour.lowlatency
patching file debian.master/config/powerpc/config.common.powerpc
patching file debian.master/config/ppc64el/config.common.ppc64el
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 109 (offset 1 line).
patching file drivers/media/usb/uvc/uvcvideo.h
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 594 (offset 15 lines).
cp: no se sobreescribirá el fichero «./.config» recién creado con «/usr/src/linux-headers-4.4.4-040404-generic/.config»

Thanks

ddiakopoulos, 2016-03-19:
@javcursor One of the first lines in the dmesg is this, showing that the driver hasn't been patched.

  `uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71`

Thanks for attaching the output of the patching log. I believe this is identical to the error found here: https://github.com/intel-ros/realsense/issues/12#issuecomment-198079546

This is where it's failing: https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-uvcvideo-4.4.sh#L50. You'll need manually execute the rest of the steps in a separate shell in order to complete the patch process. Unfortunately you'll need to unwind some of the variables like KBASE, but you could run the script once again and add some print statements to see what they are if you are having trouble.

javcursor, 2016-03-19:
Problem solved, many thanks for your time.

prarobo, 2016-03-21:
I solved my problem by using the uvcvideo.ko file generated from a different identical system. I just copied and pasted this into the appropriate location. Do you see any caveats in this approach @ddiakopoulos ? Will I run into problems if I do this for several robots?

ddiakopoulos, 2016-03-21:
@prarobo should be fine so long as the `uname r` strings are **absolutely **identical. Minor kernel patch versions will cause the kernel to reject the ko. Otherwise you're fine!

bsbc, 2016-03-22:
I had the same problem. Fixed it by deleting lines 50 & 51 in the [patch_uvcvideo-4.4.sh script](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-uvcvideo-4.4.sh#L50) and used the following instead:
CONFIG_LOCATION=/usr/src/linux-headers-$(uname -r)

ddiakopoulos, 2016-03-26:
@bsbc I integrated this change in a new commit: 76b33a02640e6adab496238574af07be6855dbe1

closed 2016-03-26 by ddiakopoulos

jvgomez, 2016-04-06:
Hi, I was having some problems with the camera but thanks to this issue and to [this other issue](https://github.com/intel-ros/realsense/issues/12) I was able to fix it. However, I wonder why the changes that fix the problem in the script have been undone.

Thanks!

Thomas00010111, 2016-05-23:
When you remove the lines 50 & 51 then the two files are not copied i.e. missing.
CONFIG_LOCATION=/usr/src/linux-headers-$(uname -r) 
Is needed, though.

Thomas

niccarey, 2016-06-10:
Hi, I'm running into a similar problem with the SR300: modprobe reveals repeated warnings involving uvcvideo, and if I try and start up a window with a stream (eg. using cpp-tutorial-2-streams) neither colour, depth or IR come up at all (not even a black window*). I'm using the Video4Linux backend, to minimize potential issues. I tried both the 4.4 kernel script and the mainline script patch, on clean installs, but had the same problem on both. 

Since it seems related, I thought I'd post here as it could well be a patch error, given the uvcvideo warnings. However none of the fixes listed here seem to work - let me know if it's more appropriate to start a new issue.

Changing the patch script as suggested above simply throws up more errors. I ran the SR300 live unit tests script and it reported no problems ("all tests passed), however a further modprobe of uvc after running the script showed multiple "uvcvideo: Non-zero status" and "uvcvideo: Failed to query UVC control" errors (modprobe output copied below, sorry for the length).

* I suppose there is a slight chance this is actually a glfw problem, since the window output seems problematic in a different way to that reported here and elsewhere. However I ran a short glfw test program to check this, and had no issues.

I should note that I have previously set up this same SR300 using identical hardware and an identical installation process, and it worked fine (still does!). Completely baffled at this point.

`[  213.505694] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  213.507160] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.1/2-2.1:1.0/input/input16
[  213.508672] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[  213.508677] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
[  213.508750] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  213.509571] uvcvideo: Unable to create debugfs 2-3 directory.
[  213.509727] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.1/2-2.1:1.2/input/input17
[  213.510639] usbcore: registered new interface driver uvcvideo
[  213.510642] USB Video Class driver (1.1.1)
[  221.259752] uvcvideo: Non-zero status (-71) in video completion handler.
[  223.558650] uvcvideo: Non-zero status (-71) in video completion handler.
[  237.014474] uvcvideo: Non-zero status (-71) in video completion handler.
[  239.620868] uvcvideo: Non-zero status (-71) in video completion handler.
[  241.937123] uvcvideo: Non-zero status (-71) in video completion handler.
[  245.509028] uvcvideo: Non-zero status (-71) in video completion handler.
[  249.089001] uvcvideo: Non-zero status (-71) in video completion handler.
[  296.357967] uvcvideo: Non-zero status (-71) in video completion handler.
[  298.981624] uvcvideo: Non-zero status (-71) in video completion handler.
[  301.293752] uvcvideo: Non-zero status (-71) in video completion handler.
[  304.878369] uvcvideo: Non-zero status (-71) in video completion handler.
[  308.452756] uvcvideo: Non-zero status (-71) in video completion handler.
[  310.783042] uvcvideo: Non-zero status (-71) in video completion handler.
[  317.040678] uvcvideo: Non-zero status (-71) in video completion handler.
[  323.271887] uvcvideo: Non-zero status (-71) in video completion handler.
[  329.518868] uvcvideo: Non-zero status (-71) in video completion handler.
[  335.655802] uvcvideo: Non-zero status (-71) in video completion handler.
[  341.829035] uvcvideo: Non-zero status (-71) in video completion handler.
[  341.834817] uvcvideo: Non-zero status (-71) in video completion handler.
[  344.605358] uvcvideo: Non-zero status (-71) in video completion handler.
[  344.611159] uvcvideo: Non-zero status (-71) in video completion handler.
[  347.050978] uvcvideo: Non-zero status (-71) in video completion handler.
[  347.058422] uvcvideo: Non-zero status (-71) in video completion handler.
[  349.509821] uvcvideo: Non-zero status (-71) in video completion handler.
[  349.520519] uvcvideo: Non-zero status (-71) in video completion handler.
[  351.976343] uvcvideo: Non-zero status (-71) in video completion handler.
[  351.982767] uvcvideo: Non-zero status (-71) in video completion handler.
[  354.405937] uvcvideo: Non-zero status (-71) in video completion handler.
[  354.411688] uvcvideo: Non-zero status (-71) in video completion handler.
[  356.900522] uvcvideo: Non-zero status (-71) in video completion handler.
[  360.453683] uvcvideo: Non-zero status (-71) in video completion handler.
[  362.344638] uvcvideo: Non-zero status (-71) in video completion handler.
[  363.973986] uvcvideo: Non-zero status (-71) in video completion handler.
[  371.600058] uvcvideo: Non-zero status (-71) in video completion handler.
[  371.607259] uvcvideo: Non-zero status (-71) in video completion handler.
[  374.130385] uvcvideo: Non-zero status (-71) in video completion handler.
[  374.140093] uvcvideo: Non-zero status (-71) in video completion handler.
[  385.937575] uvcvideo: Non-zero status (-71) in video completion handler.
[  385.944819] uvcvideo: Non-zero status (-71) in video completion handler.
[  386.957043] uvcvideo: Non-zero status (-71) in video completion handler.
[  386.963308] uvcvideo: Non-zero status (-71) in video completion handler.
[  398.282402] uvcvideo: Non-zero status (-71) in video completion handler.
[... cut ...]
[  408.068448] uvcvideo: Non-zero status (-71) in video completion handler.
[  410.359062] uvcvideo: Non-zero status (-71) in video completion handler.
[  410.366652] uvcvideo: Non-zero status (-71) in video completion handler.
[  412.788960] uvcvideo: Non-zero status (-71) in video completion handler.
[  412.797656] uvcvideo: Non-zero status (-71) in video completion handler.
[  415.275198] uvcvideo: Non-zero status (-71) in video completion handler.
[  415.411975] uvcvideo: Non-zero status (-71) in video completion handler.`

ddiakopoulos, 2016-06-10:
For whatever reason, the patch isn't being applied correctly on your system (unknown video formats are a telltale sign). Follow the script closely, and don't forget to `sudo modprobe uvcvideo` at the end

niccarey, 2016-06-10:
I tried again on a completely clean Ubuntu 14.04 install, to make sure I hadn't made an error in the process. The last few lines of dmesg after patch application and modprobe is

[  241.789686] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel
[  241.790724] usbcore: registered new interface driver uvcvideo
[  241.790729] USB Video Class driver (1.1.1)

niccarey, 2016-06-10:
update: tried on a completely clean Ubuntu 14.04 install on a different machine - no disturbing messages from uvcvideo in dmesg, and the camera seems to be working. No doubt I did something careless during the patch application on the previous system, I will roll back (again!) and see if I can get it working.

niccarey, 2016-06-10:
Further update: re-installing Ubuntu and running the installation procedure **exactly** as on the first and third machine (which both work fine) still produces no useable image.

After applying the patch, the modprobe reveals nothing unusual (except the module verification failure, but this seems to happen whether or not the patch succeeds as I get the same message on other machines)

[  351.929478] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel
[  351.931636] usbcore: registered new interface driver uvcvideo
[  351.931643] USB Video Class driver (1.1.1)
(..)

After plugging in the realsense and running cpp-tutorials-2-streams, I get nothing at all in the glfw window and running modprobe produces the following dmesg output:

[  662.018483] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[  662.018490] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
[  662.018561] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  662.019470] uvcvideo: Unable to create debugfs 2-3 directory.
[  662.019700] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.1/2-2.1:1.2/input/input17

So at this point it looks like there's something more fundamental going on, and the problem is not with the patch itself. I am getting a weirdly persistent grub screen on boot, which might mean there's another kernel installed, although I am using the same installation USB as on the working machines, and when I reinstalled ubuntu I erased the disk. I will check the partitions and try again I suppose.

l3s777, 2016-08-23:
when running all patches, etc... I got the same issue in msg:
[ 351.929478] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel

ravenblackx, 2016-09-27:
I too have the error
uvcvideo: Non-zero status (-71) in video completion handler.
in dmesg. I also had an issue with /dev/video0 permissions (remedied by adding the user to the group 'video').
Without the permissions fix, the error behavior was "no device detected". After the permissions fix, I get the -71 error in dmesg when I start up any camera-using software (eg. cpp-headless), and the infrared and depth images appear to be all zeroes.

Perhaps the most interesting thing here is that I had been using the camera fine the previous day, then after a reboot of the system both of these problems arose. There may have been an apt-get upgrade before the reboot - is that expected to break the driver?

ravenblackx, 2016-09-27:
Rerunning the patch script did not fix the issue described above, but rerunning it, telling it 'yes' when it asks a question about the patch already being applied that means "do you want to unpatch instead of patch" (I forget how it's phrased, but not very helpfully!), then rerunning it a second time, allows the script to run to completion, which fixed my issue.
I think the apt-get upgrade was the cause of the problem - it continues to work after a subsequent reboot.

l3s777, 2016-09-28:
mine, it's still not working!

mpowell326, 2016-12-19:
I also had the issue of unknown video format after running the patching script and loading the uvcvideo module.

It turns out that the script was not running successfully for me even though no errors were shown. This was due to the 'realsense-camera-format patch' having a relative address (line 47). Re-running the script from the 'librealsense' dir rather than 'librealsense/scripts' fixed this issue for me.

mpowell326, 2016-12-19:
I also had the issue of unknown video format after running the patching script and loading the uvcvideo module.

It turns out that the script was not running successfully for me even though no errors were shown. This was due to the 'realsense-camera-format patch' having a relative address (line 47) and was not being found. Re-running the script from the 'librealsense' dir rather than 'librealsense/scripts' fixed this issue for me.

boulabiar, 2017-01-13:
I did everything from the begining in another folder and I always get this when trying to run any example:

rs_error was raised when calling rs_create_context(api_version:11201):
    UVCIOC_CTRL_QUERY:UVC_GET_CUR error 56, Invalid request code