realsenseai/librealsense #71: VIDIOC_DQBUF error 19, No such device

Nikvand, 2016-03-14:
| Camera Model | F200 | 
| Firmware Version | 2.60.0.0  | 
| Operating System & Version |  Ubuntu 14.04 LTS |
| Kernel Version (Linux Only)  |  4.4.0-040400 generic |
| Build System |  Makefile |  

Hi I am getting the following error after installation:

~/librealsense/bin$ ./cpp-headless
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense F200
    Serial number: 01FFFFFF8150165706
    Firmware version: 2.60.0.0
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)

No errors in dmesg | grep error !

ddiakopoulos, 2016-03-14:
Hi @Nikvand - VIDIOC_DQBUF problems are tricky to debug. I would first start with trying the camera through a _powered_ USB 3.0 hub, or on a different USB port. If the camera enumerates and gives you data about its firmware that is a good sign, but we've seen issues when devices attempt to stream camera data and the USB bus/port isn't supplying enough power.

Nikvand, 2016-03-14:
Hi @ddiakopoulos Yes I have noticed that the same firmware works fine on my other computer. Unfortunately this is the computer I use at work and has only one USB 3.0 port.

ddiakopoulos, 2016-03-14:
@Nikvand definitely test through a powered USB hub first. Would be a bummer if it was instead the USB host controller on your work machine.

Nikvand, 2016-03-14:
@ddiakopoulos Thanks for your replies. When I run the cpp-restart it shows me the BGR picture with no IR stream or Depth. I have had many problems on ubuntu drivers for this particular machine so there's a chance it is the USB host controller causing the problem as you said.

ddiakopoulos, 2016-03-17:
@Nikvand You could try reducing the FPS and resolution as part of one of the example apps (cpp-capture) and check if that improves streaming reliability. Also only enable color/depth and don't stress the system by attempting to capture the IR. Hope that helps!

oliviakim95, 2016-03-24:
@Nikvand I was getting this error too and using a powered usb 3.0 hub did the trick

ddiakopoulos, 2016-03-26:
@Nikvand closing this out since I don't think I can offer any additional insight on your specific hardware configuration.

closed 2016-03-26 by ddiakopoulos

ckim337, 2016-05-05:
@ddiakopoulos Is there any specific powered hub required for SR300? ( I have used the power hub from amazon that gives 4 Port USB 3.0 Hub with 5V/2.5A power adapter, but still not working. Maybe this hub is not strong enough. "http://www.amazon.com/AmazonBasics-Port-2-5A-power-adapter/dp/B00DQFGH80")
@Nikvand  Can you please provide us the information about the powered hub that you used?

mga0, 2016-11-16:
I also had this problem in some examplefiles. 

>  'std::runtime_error' what():  VIDIOC_DQBUF error 19, No such device  [1]    8356 abort (core dumped)

We're using R200 (_Firmware version: 1.0.72.10_) and RS300 (_Firmware version: 3.21.0.0_). 
R200 crashes after a few seconds, and RS300 crashes when you move your hands towards the camera. 

In the end we could track it down to the following:
**The R200** with this exception could be solved by using **an active USB3.0 Hub**. 
But having the **RS300** connected on the hub produced the same exception . Here we went back to the **computer's USB3.0 port** and the problem was gone. 

I assume, the R200 consumes more energy and needs stable power supply. Since it is claimed that the RS300 consumes less energy, maybe the connection encounters a bandwidth problem, which was solved by attaching the device directly.