realsenseai/librealsense #72: LIBUSB_ERROR_PIPE

pancx, 2016-03-15:
| Required Info  |  When run some examples,like restart, it turns out that
`RealSense error calling rs_start_device(device:0x10f5678):
    uvc_set_ctrl(...) returned LIBUSB_ERROR_PIPE` 
So, where may the mistake be?|
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.71..... | 
| Operating System & Version |  Ubuntu14.04 |
| Kernel Version (Linux Only)  | 4.4  |
| Build System |  QT, Makefile |

pancx, 2016-03-15:
And this 
`rs.warn: Camera head contents version != 12, data may be missing/incorrect
`
often occurred.

ddiakopoulos, 2016-03-15:
Hi @pancx. The warning is insubstantial so don't worry about it. In the other case, I recommend upgrading your firmware as outlined here: https://github.com/IntelRealSense/librealsense/issues/70#issuecomment-196398678

pancx, 2016-03-17:
@ddiakopoulos Thanks for your reply. 
I have patched the uvcvideo again according to #51 , and no errors appeared.
 When run the cpp-config-ui, the color, depth, infrared, infrared2 could display respectively, but if let it display all four images, it well quit immediately.
So have I installed librealsense completely and successfully?
If not, why single image is normally displaying?

ddiakopoulos, 2016-03-17:
If you are able to view each of the color, depth and IR streams, then librealsense has been correctly installed. 

For your new issue, it could be USB 3.0 bandwidth. Best to experiment with different combinations of enabled streams, FPS, and resolution. cpp-config-ui doesn't permit you to play with the FPS and resolution settings, so you will need to modify one of the examples.

ddiakopoulos, 2016-03-26:
Closing for lack of new info. Feel free to update and I will re-open if required.

closed 2016-03-26 by ddiakopoulos

pancx, 2016-04-08:
@ddiakopoulos 
The CPU of my desktop is i7-2600k. Is it the reason why the window would quit when display four images together?
I have installed librealsense on Minnowboard Max. To my surprise, it works normally.