realsenseai/librealsense #75: Insufficient permissions in install_dependencies-4.4.sh

archielee, 2016-03-18:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.4  |
| Build System |  VS2013, XCode or Makefile | 

File requires installing via apt-get and updating grub, all of which require root privileges. Not sure if it's best practice to include them in the script which is why I haven't made a pull request. I'm writing a script to automate the install so I'm calling that script and root permissions aren't given or asked for.

ddiakopoulos, 2016-03-18:
Apt-get is well-known to require sudo? If you find particular commands in the script that require an additional sudo qualifier, let us know!

archielee, 2016-03-18:
Exactly my point. The install_dependencies script is calling a lot of apt-get commands without root privileges. I'll make a pull request to fix that.

ddiakopoulos, 2016-03-19:
@archielee thanks for the pull. Closing this issue out and will merge the pull in the next little while.

closed 2016-03-19 by ddiakopoulos

archielee, 2016-03-19:
Thanks a lot!