realsenseai/librealsense #77: Noisy depth close to camera

connerbrooks, 2016-03-18:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.06  | 
| Operating System & Version |  Windows 10, 10586.164 |
| Build System |  VS2013 | 

Depth data close camera is extremely noisy and inconsistent as seen below. Is this the expected behavior? If so, is there a way to clip data to a near and far plane from within librealsense?

I could pre-process the frames in my application, but I think this is functionality that belongs in the library.

[image]

ddiakopoulos, 2016-03-18:
Hmm, librealsense is specifically for camera capture. Image post-processing isn't in the scope of the library. For R200 data, there is a hardware control to set the minimum depth distance. You can also try setting one of the depth control presets to 3, 4, or 5 for higher outlier rejection.

connerbrooks, 2016-03-18:
The problem is that this isn't easy to filter after the fact. As the data is not at an extreme but rather shows up as good depth data, I would think this should just return no depth data.

[image]

ddiakopoulos, 2016-03-18:
The data you're seeing is explicitly generated as part of algorithm on the ASIC. Not sure what your scene actually looks like, but I am suspicious at that amount of noise. Could be a miscalibrated unit? 

Try with a setting of 5:

`rs_apply_depth_control_preset(dev, 5);`

connerbrooks, 2016-03-19:
I don't think my unit miscalibrated, as were seeing data like this with the 4 units we have at our lab. It only occurs when objects are too close.

Increasing the depth control present to 5 definitely helps reduce the noise I'm seeing.

ddiakopoulos, 2016-03-19:
Ok, great. In any case, a naive plane clip on your end should help the remainder. Otherwise, you can also clamp on the hardware side to an application-specific minimum distance using the `r200_depth_clamp_min` option.

closed 2016-03-19 by ddiakopoulos

l3s777, 2016-08-10:
hey guys!
I also got issues with the quality in the depth image... it is very noisy!