realsenseai/librealsense #78: Color PNG appears with lower intensity

Goldesel23, 2016-03-19:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   1.0.72.04  | 
| Operating System & Version | W10  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013| 

Hi, 

When trying to save the PCloud, depth, ir and color image using librealsense the color image and IR image appear with low intensity... I guess it's a code problem, because in the examples the images work fine.

Can anyone explain why this happens? Here's the sample code:

`// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2015 Intel Corporation. All Rights Reserved.

///////////////////////////////////////////////////////
// librealsense tutorial #3 - Point cloud generation //
///////////////////////////////////////////////////////

// First include the librealsense C++ header file
#include <librealsense/rs.hpp>
#include <cstdio>
#include <vector>

#define STB_IMAGE_WRITE_IMPLEMENTATION
#include "third_party/stb_image_write.h"

// Also include GLFW to allow for graphical display
#define GLFW_INCLUDE_GLU
#include <GLFW/glfw3.h>
#include <fstream>

using namespace std;

double yaw, pitch, lastX, lastY; int ml;
static void on_mouse_button(GLFWwindow * win, int button, int action, int mods)
{
    if(button == GLFW_MOUSE_BUTTON_LEFT) ml = action == GLFW_PRESS;
}
static double clamp(double val, double lo, double hi) { return val < lo ? lo : val > hi ? hi : val; }
static void on_cursor_pos(GLFWwindow * win, double x, double y)
{
    if(ml)
    {
        yaw = clamp(yaw - (x - lastX), -120, 120);
        pitch = clamp(pitch + (y - lastY), -80, 80);
    }
    lastX = x;
    lastY = y;
}

void normalize_depth_to_rgb(uint8_t rgb_image[640 * 480 * 3], const uint16_t depth_image [], int width, int height)
{
	for (int i = 0; i < width * height; ++i)
	{
		if (auto d = depth_image[i])
		{
			uint8_t v = d * 255 / std::numeric_limits<uint16_t>::max();
			rgb_image[i * 3 + 0] = 255 - v;
			rgb_image[i * 3 + 1] = 255 - v;
			rgb_image[i * 3 + 2] = 255 - v;
		}
		else
		{			
			rgb_image[i * 3 + 0] = 0;
			rgb_image[i * 3 + 1] = 0;
			rgb_image[i * 3 + 2] = 0;
		}
	}	
}

int main() try
{
    // Turn on logging. We can separately enable logging to console or to file, and use different severity filters for each.
    rs::log_to_console(rs::log_severity::warn);
    //rs::log_to_file(rs::log_severity::debug, "librealsense.log");

    // Create a context object. This object owns the handles to all connected realsense devices.
    rs::context ctx;
    printf("There are %d connected RealSense devices.\n", ctx.get_device_count());
    if(ctx.get_device_count() == 0) return EXIT_FAILURE;

    // This tutorial will access only a single device, but it is trivial to extend to multiple devices
    rs::device * dev = ctx.get_device(0);
    printf("\nUsing device 0, an %s\n", dev->get_name());
    printf("    Serial number: %s\n", dev->get_serial());
    printf("    Firmware version: %s\n", dev->get_firmware_version());

    // Configure depth and color to run with the device's preferred settings
    dev->enable_stream(rs::stream::depth, rs::preset::best_quality);
    dev->enable_stream(rs::stream::color, rs::preset::best_quality);
	dev->enable_stream(rs::stream::infrared, rs::preset::best_quality);

	// Infrared 2 will only work with R200 model.
	bool ir2_on = false;
	try{ dev->enable_stream(rs::stream::infrared2, rs::preset::best_quality);}
	catch (...) { 
		ir2_on = true;
		printf("Device does not provide infrared2 stream.\n"); }

	// Start the device
    dev->start();

    // Open a GLFW window to display our output
    //glfwInit();
    //GLFWwindow * win = glfwCreateWindow(1280, 960, "librealsense tutorial #3", nullptr, nullptr);
    /*glfwSetCursorPosCallback(win, on_cursor_pos);
    glfwSetMouseButtonCallback(win, on_mouse_button);
    glfwMakeContextCurrent(win);*/
    //while(!glfwWindowShouldClose(win))
	while (true)
	{
        // Wait for new frame data
       // glfwPollEvents();
        dev->wait_for_frames();

        // Retrieve our images
        const uint16_t * depth_image = (const uint16_t *)dev->get_frame_data(rs::stream::depth);
        const uint8_t * color_image = (const uint8_t *)dev->get_frame_data(rs::stream::color);
		const uint8_t * ir_frame = reinterpret_cast<const uint8_t *>(dev->get_frame_data(rs::stream::infrared));
		if (ir2_on){
			const uint8_t * ir_frame2 = reinterpret_cast<const uint8_t *>(dev->get_frame_data(rs::stream::infrared2));
		}

		const int width = 640;
		const int height = 480;

		

        // Retrieve camera parameters for mapping between depth and color
        rs::intrinsics depth_intrin = dev->get_stream_intrinsics(rs::stream::depth);
        rs::extrinsics depth_to_color = dev->get_extrinsics(rs::stream::depth, rs::stream::color);
        rs::intrinsics color_intrin = dev->get_stream_intrinsics(rs::stream::color);
        float scale = dev->get_depth_scale();

		std::vector<uint8_t> coloredDepth(depth_intrin.width * depth_intrin.height * 3);
		normalize_depth_to_rgb(coloredDepth.data(), depth_image, depth_intrin.width, depth_intrin.height);
		stbi_write_png("cpp-headless-output-depth.png", depth_intrin.width, depth_intrin.height, 3, coloredDepth.data(), 3 * depth_intrin.width);
		stbi_write_png("cpp-headless-output-rgb.png", color_intrin.width, color_intrin.height, 3, color_image, 3 * color_intrin.width);
		stbi_write_png("cpp-headless-output-ir.png", depth_intrin.width, depth_intrin.height, 1, ir_frame, depth_intrin.width);

        // Set up a perspective transform in a space that we can rotate by clicking and dragging the mouse
        /*glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT);
        glMatrixMode(GL_PROJECTION);
        glLoadIdentity();
        gluPerspective(60, (float)1280/960, 0.01f, 20.0f);
        glMatrixMode(GL_MODELVIEW);
        glLoadIdentity();
        gluLookAt(0,0,0, 0,0,1, 0,-1,0);
        glTranslatef(0,0,+0.5f);
        glRotated(pitch, 1, 0, 0);
        glRotated(yaw, 0, 1, 0);
        glTranslatef(0,0,-0.5f);*/

        // We will render our depth data as a set of points in 3D space
        /*glPointSize(2);
        glEnable(GL_DEPTH_TEST);
        glBegin(GL_POINTS);*/

		ofstream myfile;
		myfile.open("pCloud_noGL.txt");

        for(int dy=0; dy<depth_intrin.height; ++dy)
        {
            for(int dx=0; dx<depth_intrin.width; ++dx)
            {
                // Retrieve the 16-bit depth value and map it into a depth in meters
                uint16_t depth_value = depth_image[dy * depth_intrin.width + dx];
                float depth_in_meters = depth_value * scale;

                // Skip over pixels with a depth value of zero, which is used to indicate no data
                if(depth_value == 0) continue;

                // Map from pixel coordinates in the depth image to pixel coordinates in the color image
                rs::float2 depth_pixel = {(float)dx, (float)dy};
                rs::float3 depth_point = depth_intrin.deproject(depth_pixel, depth_in_meters);
                rs::float3 color_point = depth_to_color.transform(depth_point);
                rs::float2 color_pixel = color_intrin.project(color_point);

                // Use the color from the nearest color pixel, or pure white if this point falls outside the color image
                /*const int cx = (int)std::round(color_pixel.x), cy = (int)std::round(color_pixel.y);
                if(cx < 0 || cy < 0 || cx >= color_intrin.width || cy >= color_intrin.height)
                {
                    glColor3ub(255, 255, 255);
                }
                else
                {
                    glColor3ubv(color_image + (cy * color_intrin.width + cx) * 3);
                }*/

                // Emit a vertex at the 3D location of this depth pixel
                //glVertex3f(depth_point.x, depth_point.y, depth_point.z);
				myfile << depth_point.x << '\t' << depth_point.y << '\t' << depth_point.z << endl;
				

            }
        }
        //glEnd();

        //glfwSwapBuffers(win);

		/*if (ir2_on){
			stbi_write_png("cpp-headless-output-ir2.png", width, height, 1, ir_frame2, width);
		}*/
		

		return EXIT_SUCCESS;
    }
    
    return EXIT_SUCCESS;
}
catch(const rs::error & e)
{
    // Method calls against librealsense objects may throw exceptions of type rs::error
    printf("rs::error was thrown when calling %s(%s):\n", e.get_failed_function().c_str(), e.get_failed_args().c_str());
    printf("    %s\n", e.what());
    return EXIT_FAILURE;
}`

If anyone knows the problem, I would be very thankfull.

Thank you for your time, and great library!

bobdavies2000, 2016-03-19:
What happened to the include statements?  Did gmail clip the file names or
was your original missing the filename after #include?

ddiakopoulos, 2016-03-19:
@Goldesel23 so this looks like a slightly modified version of the cpp-headless example. PNG as a file format doesn't usually support 16 bit data (in the case of RealSense depthmaps). There's a function here normalize_depth_to_rgb() which normalizes the 16 bit depth data into an 8 bit 0-255 RGB range, explaining the different colors you see.  

This example was meant to be used for debugging only since the write function is a lossy operation. You can still write 16 bit values into a PNG, but they can't be used for visualization by most image viewing apps.

Goldesel23, 2016-03-19:
@ddiakopoulos I already were able to get the correct png samples for depth, the problem is the color data that in the display appears with the right luminosity and when i save it, it loses luminosity. I already got samples with good quality, but when obtaining the point cloud with this code, it seems to create an image with less luminosity... 

@bobdavies2000 it remove them just to show the main code

ddiakopoulos, 2016-03-20:
@Goldesel23 it might be helpful to post pictures (you can attach them directly in a Github issue).

Goldesel23, 2016-03-20:
@ddiakopoulos  These are the images with that code and with the sdk for the color samples. 

[image]

[image]

ddiakopoulos, 2016-03-20:
@Goldesel23 can you stream color using our example app cpp-capture?

Goldesel23, 2016-03-20:
[image]

Yes i can! But when acessing the Point cloud with that code the images loose intensity. I was trying to get PCloud, depth, IR and color samples for a database acquisition. So i suspected it was a code problem...

Goldesel23, 2016-03-20:
I think i Found the problem. I didn't include the initial 30 frames to /to give autoexposure, etc. a chance to settle so the image was with lower intensity. 

After including a for `(int i = 0; i < 30; ++i) dev->wait_for_frames();` the images captured were fine!

closed 2016-03-20 by Goldesel23