realsenseai/librealsense #79: OpenCV integration with librealsense

Goldesel23, 2016-03-20:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |1.0.72.04   | 
| Operating System & Version |  W10  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013 | 

Sorry to open a new issue, but this is a different problem. 

When i was trying to add the openCV spreadsheet with librealsense the samples stop working. Has anyone tried to integrate the two libraries? 

I wanted to process some of the acquisitions for some applications, while also using librealsense. 

Thank you for your time!

bobdavies2000, 2016-03-20:
I am using librealsense with OpenCV on both Windows and Linux.  Should work
fine for you.

ddiakopoulos, 2016-03-20:
Yes, this is a supported use case and I believe most users of librealsense integrate both the library and OpenCV into the same application. Unfortunately providing help on this topic is outside the scope of Github issues.

closed 2016-03-20 by ddiakopoulos

Goldesel23, 2016-03-20:
@ddiakopoulos when I work with a project with openCV or librealsense, they work fine, with their respective worksheets. But when I use the two simoultaneosly the project stops working with openCV or with librealsense (depending on the order of the dependencies of VS2013). 

For you it was straightfoward the integration with VS2013?

bobdavies2000, 2016-03-21:
Try this sample VS2013 project that was setup to use OpenCV 3.1 from
nuget.  It will ignore your locally installed OpenCV and should just run:

https://github.com/bobdavies2000/handSample.git

It was unnecessary for the project to use a DLL but it was targeted to a
project that was ultimately going to run with Unity.

Goldesel23, 2016-03-21:
I have tried to implement your project before, but it fails due to a memory problem. I'll keep trying to implement a project that is able to integrate both libraries. Thank you for your time!

Goldesel23, 2016-03-21:
So i guess i was trying to integrate x86 openCV libs with x64 librealsense libs and that generated the conflits, resolved the problem. Thanks for the help. 

And great library again!

ddiakopoulos, 2016-03-21:
@Goldesel23 :+1: