realsenseai/librealsense #80: librealsense always claims all connected cameras

xqms, 2016-03-21:
Hi,

in our application it would make sense to have two separate processes using two separate cameras. This is currently not possible with librealsense, since it always opens all cameras such that the second process gets the error:

```
RealSense error calling rs_create_context(api_version:4):
    libusb_claim_interface(...) returned LIBUSB_ERROR_BUSY
```

I understand that this is not easy to solve, since it seems that librealsense actually needs to claim the usb interface in order to determine the serial number of the device.
Releasing the interface between the query and the actual usage of the device could allow a second process to claim the unused device.

Ideally, the RealSense devices would export their serial number as the USB "iSerial" attribute - then userspace applications could see that as sysfs attribute and could quickly locate the correct device without communicating with each one. However, iSerial is always 0 for our RealSense SR300 devices (see `lsusb -v` output).

Cheers,
  Max

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.10.10.0 | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile |

ddiakopoulos, 2016-03-21:
We've seen blank serial numbers before and it's hit-or-miss depending on the version of the firmware. 

It might be easier to solve than you think. We will need to keep track of claimed interfaces on the context (or per-device), and release unused ones immediately after a call to start streaming.

Thanks for your report... multi-process isn't something we've focused on in the past, but this should be supported use case. Will update this issue when we've had a chance to look.

emulvaney-aevena, 2016-03-23:
I've noticed this too, with the libUVC back-end.  If you find the code that does the device enumeration, you can wrap a quick try/catch around the device probing/creation, and if something goes wrong just print the exception and skip over it.  So, it should be able to skip over cameras that are already used by other processes.

xqms, 2016-03-23:
@emulvaney-aevena That's actually not enough - it seems librealsense keeps all interfaces claimed, even if they are not used. So a second process can't find *any* unclaimed device.
I'm only speaking for the V4L2 backend, though, things might be different with libUVC.

ddiakopoulos, 2016-05-29:
@xqms @emulvaney-aevena our ROS counterparts at Intel (https://github.com/intel-ros/realsense) recently needed multi-process access so multiple nodes could own different cameras. A pull request was recently merged into master (https://github.com/IntelRealSense/librealsense/pull/145) that should help with this issue.

closed 2016-06-25 by ddiakopoulos