realsenseai/librealsense #81: Permissions issue? `uvc_open2(...) returned Access denied`

peci1, 2016-03-21:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.0.64  | 
| Operating System & Version | Ubuntu 14.04.4 64bit |
| Kernel Version (Linux Only)  |  4.5 patched |
| Build System |  Makefile | 

I built librealsense using the installation guide with the UVC backend (I assume this is needed to get ROS support?).

Running the demo apps from within QT creator works.

When I try running them from the command line, I get this error:

    RealSense error calling rs_create_context(api_version:4):
    uvc_open2(...) returned Access denied

Running the app as root works around this issue, but I'd like to know why this happens. It also makes ROS usage difficult, because ROS is not intended to run as root.

    $ ll /dev/video*
    crw-rw----+ 1 root video 81, 0 bře 21 16:42 /dev/video0
    crw-rw----+ 1 root video 81, 1 bře 21 16:42 /dev/video1

ddiakopoulos, 2016-03-21:
Lots of little issues but luckily this shouldn't be too difficult to fix:

1. The `libuvc` backend isn't supported on Linux due to an unresolved issue with releasing devices.
2. librealsense shouldn't require root permissions. Rather than tracking down the specific issue with the `libuvc` backend, I recommend installing the `V4L2` backend.
3. Our installation guide instructs you to install a blacklist file for libuvc which will specifically need to be removed before you can install the `V4L2` backend. This is partially our fault for not having an uninstaller script (the file in question is `uvc.conf` in `/etc/modprobe.d`)
4. Somewhere the documentation for the ROS node needs to indicate that you should have installed the `V4L2` backend.
5. Do you already have a custom kernel installed? You might need to modify our own uvcvideo patching script to handle this new kernel version (I tried to make the script future-proof so it _could_ be as easy as changing the argument from `v4.4-wily` to `v4.5-wily`

Last but not least, the 1.0.0.64 firmware is almost 2 years old! We don't have a supported firmware upgrade path on Linux systems, but installing the Intel RealSense DCM on a Windows 10 system while a camera is plugged in will flash your firmware to the latest recommended release, 1.0.72.06

After all that, your camera should be able to work perfectly in ROS using the V4L2 backend with no permissions implications.

peci1, 2016-03-21:
1) Well, the libuvc backend works for me both in QT creator and when running the programs as root. Maybe the issue has been fixed? :)

2+3) Tried it, it works :)

4) Created an issue for that: https://github.com/intel-ros/realsense/issues/16

5) Yes, I've used the newest development kernel for Ubuntu and patched it using your script. The patches applied seamlessly, so I assume it still works. Do I actually need to do this when using the V4L backend?

6) Yes, I know the firmware is old, but currently we have no Win8.1 machine at hand, so it's gonna take some time to upgrade the FW.

Thanks for your help.

closed 2016-03-21 by peci1