realsenseai/librealsense #82: uvcvideo kernel version mismatch

archielee, 2016-03-21:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | Linux  |
| Kernel Version (Linux Only)  | 4.4  |
| Build System | Makefile | 

I re-ran the installation on a clean install to see if I could consistently install it without any problems. This time around, I ran into an odd issue. The computer refused to detect my R200 then after viewing the syslog, I found the line "uvcvideo: disagrees about version of symbol module_layout". A quick google search shows a kernel mismatch (http://stackoverflow.com/questions/2720177/module-layout-version-incompatibility). Did anything change recently? If it helps, the original install failed to install gcc 4.9 and I had to rerun that script then the uvc patch.

archielee, 2016-03-21:
Issue has been solved, rebooted and re-ran the patch.

closed 2016-03-21 by archielee

cyberf1sh, 2016-11-02:
Hi, 

I'm running into the same issue, except rebooting and re-rerunning the patch is not fixing it for me. I have same configuration as archielee too. Can someone please help me with this?

archielee, 2016-11-02:
Do you have any of the errors when it fails? Syslogs, verbose output?