realsenseai/librealsense #83: 3D scanning

jmaye, 2016-03-22:
Thank you all for the nice work!
Do you plan to integrate the 3D scanning capabilities from the Windows SDK?
If not, any hints about the algorithm used (relevant paper) in case I want to implement it myself.

ddiakopoulos, 2016-03-22:
There's some internal discussion about releasing some of the SDK middleware for Linux developers (as binary releases). No final decisions on that front.

The ScenePerception module in the SDK implements a classical KinectFusion style algorithm, accelerated in OpenCL 1.2 and with custom extensions for larger spaces similar in functionality detailed in the Kintinuous paper.

PCL has a CUDA accelerated version of KinFu which works reasonably well. 

There's a few more KinFu implementations floating around on GitHub, most of the them variants on the one found in PCL. Outside of that, there's a number of state-of-the-art implementations of pose tracking + meshing that vision researchers have released in the past few years... It's a mixed bag of dependencies and licenses. InfiniTAM is probably the easiest to get up and running because someone created a librealsense wrapper, but it is for noncommercial use only. InfiniTAM optionally uses a voxel hashing backend which, if used directly, is more permissively licensed. 

https://github.com/victorprad/InfiniTAM/pull/39 (links directly to a pull request with librealsense compatibility)
https://github.com/nachtmar/VoxelHashing
https://github.com/qianyizh/ElasticReconstruction
https://github.com/mp3guy/ElasticFusion
https://github.com/mp3guy/Kintinuous
https://github.com/icg-moca/Fusion (opencl)
https://github.com/Nerei/kinfu_remake (CUDA PCL version ripped out of PCL)
https://github.com/personalrobotics/OpenChisel (needs separate pose estimation)
https://github.com/tum-vision/fastfusion (also needs pose estimation)

Hope that helps!

jmaye, 2016-03-22:
Thanks for your detailed reply!

I have tried the kinfu stuff from PCL and it indeed works quite nicely. You just need to take care of really cropping the face region when doing the face scanning, otherwise ICP fails, which makes sense as the torso-face is not rigid. I will investigate a bit more though as the result is not as good yet as what I've tried under Windows. I will also take a look at the other implementations, but I'm glad to hear I was going on the right direction by digging into kinfu.

ddiakopoulos, 2016-03-24:
Sounds good! Closing the issue but feel free to post an update if you find a good solution.

closed 2016-03-24 by ddiakopoulos