realsenseai/librealsense #84: Timeout on fast movements

ChrisN-987, 2016-03-22:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   1.0.72.06| 
| Operating System & Version |  Windows 10 Pro N |
| Build System |  VS2013| 

Hi,

I have noticed that the R200 will timeout waiting for frames if I make large movements with the camera. If I keep it still it will capture indefinitely. 

I am capturing at 15FPS 1920x1080 colour and 30FPS 640x480 depth.

Any idea on how I can fix this? In my loop I am waiting for frames and writing out every 30th frame or so to file.

bobdavies2000, 2016-03-22:
I have seen this when the USB cable moves in the socket.  If you are waving
the camera around at the end of a USB cable while doing a scan, it can stop
working.  Attach the camera to a laptop or tablet and move the whole unit.

ddiakopoulos, 2016-03-22:
This is a semi-known issue with some variants of the USB 3.0 connector on the peripheral. A little tape goes a long way!

closed 2016-03-22 by ddiakopoulos

ChrisN-987, 2016-03-22:
I did have it attached to the back of my laptop with the magnetic sticker. Same issue.

ddiakopoulos, 2016-03-22:
It's a bad USB connection, probably on the peripheral side. If you can't get it to work with a little tape or hot glue for added mechanical support, I would recommend an RMA by emailing click.support@intel.com