realsenseai/librealsense #87: Problem finding libusb when compiling librealsense

narner, 2016-03-23:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  Unknown | 
| Operating System & Version | OS X 10.11.3   |
| Kernel Version (Linux Only)  |   |
| Build System |  XCode / Makefile | 

When trying to compile the library, I'm getting an error of `No package 'libusb-1.0' found`. I had followed the [instructions in the Apple OS X installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#apple-osx-installation), which included installing `libusb` through Homebrew. Running `brew install libusb` again gives the warning `libusb-1.0.20 already installed`. 

Below is the full error output:

```
librealsense (master)> make
mkdir -p obj/libuvc
mkdir -p lib
mkdir -p bin
c++ src/f200-private.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_LIBUVC_BACKEND `pkg-config --cflags --libs libusb-1.0`  -I/usr/local/include -c -o obj/f200-private.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
c++ src/f200.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_LIBUVC_BACKEND `pkg-config --cflags --libs libusb-1.0`  -I/usr/local/include -c -o obj/f200.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
c++ src/r200-private.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_LIBUVC_BACKEND `pkg-config --cflags --libs libusb-1.0`  -I/usr/local/include -c -o obj/r200-private.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
c++ src/r200.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_LIBUVC_BACKEND `pkg-config --cflags --libs libusb-1.0`  -I/usr/local/include -c -o obj/r200.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
c++ src/uvc-libuvc.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_LIBUVC_BACKEND `pkg-config --cflags --libs libusb-1.0`  -I/usr/local/include -c -o obj/uvc-libuvc.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
c++ src/uvc-v4l2.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_LIBUVC_BACKEND `pkg-config --cflags --libs libusb-1.0`  -I/usr/local/include -c -o obj/uvc-v4l2.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
c++ src/uvc-wmf.cpp -std=c++11 -fPIC -pedantic -mssse3 -O3 -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_LIBUVC_BACKEND `pkg-config --cflags --libs libusb-1.0`  -I/usr/local/include -c -o obj/uvc-wmf.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
c++ -std=c++11 -shared obj/verify.o obj/context.o obj/device.o obj/f200-private.o obj/f200.o obj/image.o obj/log.o obj/r200-private.o obj/r200.o obj/rs.o obj/stream.o obj/sync.o obj/types.o obj/uvc-libuvc.o obj/uvc-v4l2.o obj/uvc-wmf.o obj/uvc.o obj/libuvc/ctrl.o obj/libuvc/dev.o obj/libuvc/diag.o obj/libuvc/frame.o obj/libuvc/init.o obj/libuvc/stream.o `pkg-config --cflags --libs libusb-1.0` -o lib/librealsense.so
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
Undefined symbols for architecture x86_64:
  "_libusb_alloc_transfer", referenced from:
      _uvc_open2 in dev.o
      _uvc_stream_start in stream.o
  "_libusb_attach_kernel_driver", referenced from:
      _uvc_release_if in dev.o
  "_libusb_bulk_transfer", referenced from:
      rsimpl::uvc::bulk_transfer(rsimpl::uvc::device&, unsigned char, void*, int, int*, unsigned int) in uvc-libuvc.o
  "_libusb_cancel_transfer", referenced from:
      _uvc_stream_stop in stream.o
  "_libusb_claim_interface", referenced from:
      rsimpl::uvc::claim_interface(rsimpl::uvc::device&, rsimpl::uvc::guid const&, int) in uvc-libuvc.o
      _uvc_claim_if in dev.o
[... 51 lines cut ...]
      __uvc_handle_events in init.o
  "_libusb_init", referenced from:
      _uvc_init in init.o
  "_libusb_open", referenced from:
      _uvc_get_device_descriptor in dev.o
      _uvc_open2 in dev.o
  "_libusb_ref_device", referenced from:
      _uvc_ref_device in dev.o
  "_libusb_release_interface", referenced from:
      rsimpl::uvc::device::~device() in uvc-libuvc.o
      _uvc_release_if in dev.o
  "_libusb_set_interface_alt_setting", referenced from:
      _uvc_release_if in dev.o
      _uvc_stream_start in stream.o
  "_libusb_submit_transfer", referenced from:
      _uvc_open2 in dev.o
      __uvc_status_callback in dev.o
      __uvc_stream_callback in stream.o
      _uvc_stream_start in stream.o
  "_libusb_unref_device", referenced from:
      _uvc_unref_device in dev.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [lib/librealsense.so] Error 1
```

ddiakopoulos, 2016-03-23:
@narner can you try with the XCode solution?

narner, 2016-03-24:
@ddiakopoulos Ahh, that seems to work!

[image]

closed 2016-03-25 by ddiakopoulos