realsenseai/librealsense #90: error when patching uvcvideo

brightlight2084, 2016-03-28:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04.04 LTS   |
| Kernel Version (Linux Only)  |  4.4.0-040400-generic |
| Build System |   | 

I extract the zip file to ~/Documents, and then I follow the instructions. When I run this instruction "./scripts/patch-uvcvideo-4.4.sh v4.4-wily", some erros are displayed. Please see below, how can I deal with it? thank you.

**# patching file debian.master/config/i386/config.flavour.lowlatency
patching file debian.master/config/powerpc/config.common.powerpc
patching file debian.master/config/ppc64el/config.common.ppc64el
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 109 (offset 1 line).
patching file drivers/media/usb/uvc/uvcvideo.h
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 594 (offset 15 lines).
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
make[3]: *** [silentoldconfig] Error 1
make[2]: *** [silentoldconfig] Error 2
make[1]: *** No rule to make target `include/config/auto.conf', needed by `scripts'.  Stop.
make: *** [__build_one_by_one] Error 2**

ddiakopoulos, 2016-03-28:
Hi @brightlight2084 sorry for the trouble, I think this was unintentionally caused by me in the latest commit on master. Can you try again with a version of this script 1 commit behind HEAD? Here is a direct link: https://github.com/IntelRealSense/librealsense/blob/d185508c1780a63da9835b9f05e1b5cb0e07bd02/scripts/patch-uvcvideo-4.4.sh

smartbeast, 2016-03-28:
Got the same issue. The script 1 commit behind HEAD is working.

brightlight2084, 2016-03-28:
Now it's working, thank you.

closed 2016-03-28 by brightlight2084

ddiakopoulos, 2016-03-29:
@ravrichard only one way to find out! :)